ros2/ros2 #1472: [ros2 humble]:very large delay in transferring images with ros2 service

shuguangzhao01, 2023-08-09:
## Bug report

**Required Info:**

- Operating System:
  - ubuntu22.04
- Installation type:
  - binaries
- Version or commit hash:
  - ros2 humble
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

```
int main(int argc, char** argv) {
    rclcpp::init(argc, argv);

    std::shared_ptr<rclcpp::Node> node =
        rclcpp::Node::make_shared("udf_test_c");
    rclcpp::Client<UdfAccessSRV>::SharedPtr client =
        node->create_client<UdfAccessSRV>("udf_access_srv");
    // rclcpp::executors::MultiThreadedExecutor executor;
    long i = 1;
    std::string file =
        "/home/marco/workspace/material/video/1080050990-1-208.mp4";
    while (i < 1000000000000000) {
        cv::VideoCapture sensor(file);
        sensor.open(file);
        if (!sensor.isOpened()) {
            // RCLCPP_INFO(rclcpp::get_logger("test"),
            //             "open sensor................");
            sensor.open(file);
        }
        cv::Mat img;
        while (sensor.read(img)) {
            i += 1;
            boost::uuids::uuid a_uuid = boost::uuids::random_generator()();
            std::string uuid_string = boost::uuids::to_string(a_uuid);
            RCLCPP_INFO(rclcpp::get_logger("test"), "[%s] get a frame....i=%d", uuid_string.c_str(), i);
            if (!img.empty()) {
                RCLCPP_INFO(rclcpp::get_logger("test"), "send this image....", uuid_string.c_str());
                auto req = std::make_shared<UdfAccessSRV::Request>();

                sensor_msgs::msg::Image _img_msg;
                std_msgs::msg::Header _header;
                cv_bridge::CvImage _cv_bridge;
                _header.stamp = node->get_clock()->now();
                _cv_bridge = cv_bridge::CvImage(
                    _header, sensor_msgs::image_encodings::BGR8, img);
                _cv_bridge.toImageMsg(_img_msg);

                req->img = _img_msg;
                req->uuid = uuid_string;

                RCLCPP_INFO(rclcpp::get_logger("rclcpp"),
                    "[%s] service_name=udf_access_srv", uuid_string.c_str());
                while (!client->wait_for_service(
                    std::chrono::milliseconds(1000))) {
                    if (!rclcpp::ok()) {
                        RCLCPP_ERROR(rclcpp::get_logger("rclcpp"),
                            "[%s] Interrupted while waiting for the "
                            "service. Exiting.", uuid_string.c_str());
                        throw std::runtime_error("Interrupted while waiting "
                            "for the service. Exiting.");
                    }
                    RCLCPP_INFO(rclcpp::get_logger("rclcpp"),
[... 150 lines cut ...]
[DEBUG] [1691565198.927586977] [rcl]: Service server taking service request
[DEBUG] [1691565198.932197639] [rcl]: Service take request succeeded: true
[INFO] [1691565198.932249026] [udf_server]: [0688780a-a38a-483d-884d-828d82d8b0c7] receive a image
[INFO] [1691565198.935974553] [udf_server]: [0688780a-a38a-483d-884d-828d82d8b0c7] ****img.width=1920,img.hight=1080,img.channel=3
[DEBUG] [1691565198.936002508] [rcl]: Sending service response
[DEBUG] [1691565201.926497360] [rcl]: Service server taking service request
[DEBUG] [1691565201.927646480] [rcl]: Service take request succeeded: true
[INFO] [1691565201.927663745] [udf_server]: [262277d5-d7ed-467a-bdf8-960b9c10ff5c] receive a image
[INFO] [1691565201.931548321] [udf_server]: [262277d5-d7ed-467a-bdf8-960b9c10ff5c] ****img.width=1920,img.hight=1080,img.channel=3
[DEBUG] [1691565201.931582847] [rcl]: Sending service response
[DEBUG] [1691565201.947481547] [rcl]: Service server taking service request
[DEBUG] [1691565201.948585700] [rcl]: Service take request succeeded: true
[INFO] [1691565201.948601542] [udf_server]: [903f83cb-1c7f-42c9-a98f-eb4726898ab6] receive a image
[INFO] [1691565201.952043025] [udf_server]: [903f83cb-1c7f-42c9-a98f-eb4726898ab6] ****img.width=1920,img.hight=1080,img.channel=3
[DEBUG] [1691565201.952080348] [rcl]: Sending service response
[DEBUG] [1691565201.958807461] [rcl]: Service server taking service request
[DEBUG] [1691565201.959840853] [rcl]: Service take request succeeded: true
[INFO] [1691565201.959858131] [udf_server]: [48c850c9-3125-4272-be22-2576a461c771] receive a image
[INFO] [1691565201.963241185] [udf_server]: [48c850c9-3125-4272-be22-2576a461c771] ****img.width=1920,img.hight=1080,img.channel=3
[DEBUG] [1691565201.963260722] [rcl]: Sending service response

#### Expected behavior
The client uses OpenCV to read the 1920*1080 video transmission image, and the server receives the image with low latency.
#### Actual behavior
At first, the image transmission delay is about 15-20 milliseconds, and after about a few minutes, the receiving delay image delay becomes larger, about 1-3 seconds, as the log above shows

fujitatomoya, 2023-08-15:
@shuguangzhao01 can you close this one since this is duplication of https://github.com/ros2/rclcpp/issues/2271?

clalancette, 2023-08-24:
Closing as a duplicate of https://github.com/ros2/rclcpp/issues/2271

closed 2023-08-24 by clalancette

shuguangzhao01, 2023-08-28:
> 
Sorry, just got back from sick leave and just saw this reply