ros2/ros2 #1476: ros2 bag record --compression-format zstd fail

nemo1121, 2023-08-18:
ros2 humble ubuntu
use command: ros2 bag record /imu /odom -p 100000 -b 524288000 --compression-mode file --compression-format zstd，create two .zstd files，but use "zstd -d ***.zstd" failed：
root@jeray:/work/rosbag2_2023_08_18-16_13_40# zstd -d rosbag2_2023_08_18-16_13_40_0.db3.zstd 
-16_13_40_0.db3.zstd : 0 MB...     -16_13_40_0.db3.zstd : **_Read error (39) : premature end_** 
root@jeray:/work/rosbag2_2023_08_18-16_13_40# ls
metadata.yaml
rosbag2_2023_08_18-16_13_40_0.db3.zstd
rosbag2_2023_08_18-16_13_40_0.db3.zstd.zst
rosbag2_2023_08_18-16_13_40_1.db3.zstd
root@jeray:/work/rosbag2_2023_08_18-16_13_40# ros2 bag info rosbag2_2023_08_18-16_13_40_0.db3.zstd

closing.

closing.
[ERROR] [1692347604.721937690] [rosbag2_storage]: No storage id specified, and no plugin found that could open URI
No plugin detected that could open file rosbag2_2023_08_18-16_13_40_0.db3.zstd

[rosbag2_2023_08_18-16_13_40_0.db3.zip](https://github.com/ros2/ros2/files/12377367/rosbag2_2023_08_18-16_13_40_0.db3.zip)

nnolas27, 2023-08-26:
It seems like you're encountering issues with decompressing and reading the `.zstd` compressed ROS 2 bag files. The error message you provided suggests that there might be an issue with the compression format or the compression process itself. Here are a few things you can check and try:

1. File Integrity: Make sure that the `.zstd` files are not corrupted. You can check their integrity by comparing their file size with the original `.db3` files or using any checksum utility.

2. Decompression Command: The command you mentioned (`zstd -d ***.zstd`) should work for decompressing `.zstd` files. Make sure that you're using the correct file name and path. You can also try using the `-f` flag to force decompression:

  
  ` zstd -d -f rosbag2_2023_08_18-16_13_40_0.db3.zstd`
 

3. ROS 2 Bag Format: ROS 2 bag files might have an internal structure that the decompression process could be sensitive to. Ensure that the files were compressed and saved correctly using the `ros2 bag record` command you provided.

4. Version Compatibility: Ensure that the version of the `zstd` utility you are using is compatible with the compression format used. If you're using an older version of `zstd`, it might not support newer compression formats.

5. File Extension: It's unusual to see `.db3.zstd` as the file extension. Usually, the file extension would be `.zstd` for compressed files. Double-check that the file extensions are correct.

6. ROS 2 Bag Info: The error message from `ros2 bag info` indicates that it cannot open the `.zstd` file. This could be due to a compatibility issue or an incorrect URI. Try providing the full path to the `.zstd` file, like:

 
   `ros2 bag info /path/to/rosbag2_2023_08_18-16_13_40_0.db3.zstd`

7. Extract and Repackage: If the decompression continues to fail, you might consider trying to extract the original data from the `.zstd` files, if possible. Then, create new bag files and compress them again to see if the issue persists.

8. ROS 2 Updates: Make sure your ROS 2 installation and related packages are up to date. Sometimes, issues can arise due to compatibility problems between different software components.

If none of the above solutions work, you might need to consider reaching out to the ROS 2 community or the maintainers of the specific tools you are using for further assistance.

clalancette, 2024-01-04:
Given the lack of response here, I'm going to close this out.  If you are still having problems, please feel free to reopen and provide more information.

closed 2024-01-04 by clalancette

JJRedmond, 2024-05-08:
The same problem with every compressed splitted bags. At least one of the files is corrupted