ros2/ros2 #1477: After importing, an error was still reported stating that there is no public key to verify

Jzow, 2023-08-20:
I really can't solve this problem

```java
jamesozw@jamesozw-vm:~$ sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys F42ED6FBAB17C654
Warning: apt-key is deprecated. Manage keyring files in trusted.gpg.d instead (see apt-key(8)).
Executing: /tmp/apt-key-gpghome.KrkBhTK1ER/gpg.1.sh --keyserver keyserver.ubuntu.com --recv-keys F42ED6FBAB17C654
gpg: key F42ED6FBAB17C654: public key "Open Robotics <info@osrfoundation.org>" imported
gpg: Total number processed: 1
gpg:               imported: 1
jamesozw@jamesozw-vm:~$ sudo apt update
Hit:1 http://archive.ubuntu.com/ubuntu jammy InRelease                    
Get:2 http://packages.ros.org/ros2/ubuntu jammy InRelease [4,682 B]       
Err:2 http://packages.ros.org/ros2/ubuntu jammy InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY F42ED6FBAB17C654
Hit:3 http://archive.ubuntu.com/ubuntu jammy-updates InRelease
Hit:4 http://archive.ubuntu.com/ubuntu jammy-backports InRelease
Hit:5 http://archive.ubuntu.com/ubuntu jammy-security InRelease
Reading package lists... Done
W: GPG error: http://packages.ros.org/ros2/ubuntu jammy InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY F42ED6FBAB17C654
E: The repository 'http://packages.ros.org/ros2/ubuntu jammy InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```

Jzow, 2023-08-20:
I've tried searching for many solutions, but I haven't been able to solve it

clalancette, 2023-08-21:
Please follow the Installation instructions for setting up Ubuntu closely: http://docs.ros.org/en/rolling/Installation/Ubuntu-Install-Debians.html .  That should resolve your issue.

If you are still having problems, please open a question on https://robotics.stackexchange.com/questions/tagged/ros, giving lots of details of your platform and what you are trying to accomplish.

closed 2023-08-21 by clalancette