ros2/ros2 #1480: Crash in  /opt/ros/humble/include/rclcpp/rclcpp/publisher.hpp:295

ferdinandfan, 2023-09-08:
## Bug report
I was using "ros2 launch" command to start to my two depth camera. however it sometimes reports an error to publish msg at the first run: "[ERROR] [astra_camera_node-6]: process has died [pid 12650, exit code -11, cmd '/home/ferdinand/my_workspace/install/astra_camera/lib/astra_camera/astra_camera_node --ros-args -r __node:=incline_camera_node -r __ns:=/camera --params-file /tmp/launch_params_s_qb1ejn --params-file /tmp/launch_params_obnfplmh']"

When I use gdb to debug my programs, shows crash in
"    at /opt/ros/humble/include/rclcpp/rclcpp/publisher.hpp:295
295	    if (!intra_process_is_enabled_) {"

Cloud it be the simultaneously access by multiple process using  intra_process_is_enabled_ that causes the crash? Would the variable std::atomis<bool> be helpful for solving the case?
**Required Info:**

- Operating System: Ubuntu 22
- Installation type: binaries
- Version or commit hash:
  - 
- DDS implementation:
  - 
- Client library (if applicable):
  - 

#### Steps to reproduce issue

```

```

#### Expected behavior

#### Actual behavior

#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

clalancette, 2023-09-08:
> Cloud it be the simultaneously access by multiple process using intra_process_is_enabled_ that causes the crash?

No, that is a boolean.  While it could in theory be corrupted, it would never cause a crash.  My guess is that the debug symbols you are using is slightly different from the library, and thus the crash is happening at a slight different point.

To further debug this, we'll need to know much more about what you are trying to run and how you are running it.  Also a full stack trace would be helpful to try to take a look at what is happening.

ferdinandfan, 2023-09-09:
This is the full stack.

#0  0x00007f2d9eb173d0 in rclcpp::Publisher<std_msgs::msg::String_<std::allocator<void> >, std::allocator<void> >::publish<std_msgs::msg::String_<std::allocator<void> > > (this=0x0, msg=...)
    at /opt/ros/humble/include/rclcpp/rclcpp/publisher.hpp:295
#1  0x00007f2d9eb0c072 in astra_camera::PointCloudXyzNode::depthCb (this=0x7f2d70962960, 
    depth_msg=std::shared_ptr<const sensor_msgs::msg::Image_<std::allocator<void> >> (use count 3, weak count 0) = {...}, 
    info_msg=std::shared_ptr<const sensor_msgs::msg::CameraInfo_<std::allocator<void> >> (use count 9, weak count 0) = {...})
    at /home/ferdinand/my_workspace/src/fcbox/fcbox_drivers/camera/astra_camera/src/point_cloud_proc/point_cloud_xyz.cpp:185
#2  0x00007f2d9eb0b6c6 in operator() (__closure=0x7f2d709724e0, 
    msg=std::shared_ptr<const sensor_msgs::msg::Image_<std::allocator<void> >> (use count 3, weak count 0) = {...}, 
    info=std::shared_ptr<const sensor_msgs::msg::CameraInfo_<std::allocator<void> >> (use count 9, weak count 0) = {...})
    at /home/ferdinand/my_workspace/src/fcbox/fcbox_drivers/camera/astra_camera/src/point_cloud_proc/point_cloud_xyz.cpp:133
#3  0x00007f2d9eb0c953 in std::__invoke_impl<void, astra_camera::PointCloudXyzNode::connectCb()::<lambda(const ConstSharedPtr&, const ConstSharedPtr&)>&, const std::shared_ptr<const sensor_msgs::msg::Image_<std::allocator<void> > >&, const std::shared_ptr<const sensor_msgs::msg::CameraInfo_<std::allocator<void> > >&>(std::__invoke_other, struct {...} &) (__f=...)
    at /usr/include/c++/10/bits/invoke.h:60
#4  0x00007f2d9eb0c80d in std::__invoke_r<void, astra_camera::PointCloudXyzNode::connectCb()::<lambda(const ConstSharedPtr&, const ConstSharedPtr&)>&, const std::shared_ptr<const sensor_msgs::msg::Image_<std::allocator<void> > >&, const std::shared_ptr<const sensor_msgs::msg::CameraInfo_<std::allocator<void> > >&>(struct {...} &) (__fn=...) at /usr/include/c++/10/bits/invoke.h:110
--Type <RET> for more, q to quit, c to continue without paging--
mage_<std::allocator<void> > >&, const std::shared_ptr<const sensor_msgs::msg::CameraInfo_<std::allocator<void> > >&), astra_camera::PointCloudXyzNode::connectCb()::<lambda(const ConstSharedPtr&, const ConstSharedPtr&)> >::_M_invoke(const std::_Any_data &, const std::shared_ptr<sensor_msgs::msg::Image_<std::allocator<void> > const> &, const std::shared_ptr<sensor_msgs::msg::CameraInfo_<std::allocator<void> > const> &) (__functor=..., __args#0=std::shared_ptr<const sensor_msgs::msg::Image_<std::allocator<void> >> (use count 3, weak count 0) = {...}, 
    __args#1=std::shared_ptr<const sensor_msgs::msg::CameraInfo_<std::allocator<void> >> (use count 9, weak count 0) = {...}) at /usr/include/c++/10/bits/std_function.h:291
#6  0x00007f2d9e1b6abc in ?? () from /opt/ros/humble/lib/x86_64-linux-gnu/libimage_transport.so
#7  0x00007f2d9e19ff2e in ?? () from /opt/ros/humble/lib/x86_64-linux-gnu/libimage_transport.so
#8  0x00007f2d9e1a135d in void message_filters::Synchronizer<message_filters::sync_policies::ExactTime<sensor_msgs::msg::Image_<std::allocator<void> >, sensor_msgs::msg::CameraInfo_<std::allocator<void> >, message_filters::NullType, message_filters::NullType, message_filters::NullType, message_filters::NullType, message_filters::NullType, message_filters::NullType, message_filters::NullType> >::cb<1>(std::tuple_element<1, std::tuple<message_filters::MessageEvent<sensor_msgs::msg::Image_<std::allocator<void> > const>, message_filters::MessageEvent<sensor_msgs::msg::CameraInfo_<std::allocator<void> > const>, message_filters::MessageEvent<message_filters::NullType const>, message_filters::MessageEvent<message_filters::NullType const>, message_filters::MessageEvent<message_filters::NullType const>, message_filters::MessageEvent<message_filters::NullType const>, message_filters::MessageEvent<message_filters::NullType const>, message_filters::MessageEvent<message_filters::NullType const>, message_filters::MessageEvent<message_filters::NullType const> > >::type const&) () from /opt/ros/humble/lib/x86_64-linux-gnu/libimage_transport.so
#9  0x00007f2d9e1adee8 in ?? () from /opt/ros/humble/lib/x86_64-linux-gnu/libimage_transport.so

#10 0x00007f2d9eb5d0b9 in message_filters::Signal1<sensor_msgs::msg::CameraInfo_<std::allocator<void> > >::call (this=0x7f2d709630a8, event=...)
    at /opt/ros/humble/include/message_filters/message_filters/signal1.h:117
#11 0x00007f2d9eb5a161 in message_filters::SimpleFilter<sensor_msgs::msg::CameraInfo_<std::allocator<void> > >::signalMessage (this=0x7f2d709630a8, event=...)
    at /opt/ros/humble/include/message_filters/message_filters/simple_filter.h:133
#12 0x00007f2d9eb570e1 in message_filters::Subscriber<sensor_msgs::msg::CameraInfo_<std::allocator<void> >, rclcpp::Node>::cb (this=0x7f2d709630a0, e=...)
    at /opt/ros/humble/include/message_filters/message_filters/subscriber.h:349
#13 0x00007f2d9eb537c0 in message_filters::Subscriber<sensor_msgs::msg::CameraInfo_<std::allocator<void> >, rclcpp::Node>::subscribe(rclcpp::Node*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rmw_qos_profile_s, rclcpp::SubscriptionOptionsWithAllocator<std::allocator<void> >)::{lambda(std::shared_ptr<sensor_msgs::msg::CameraInfo_<std::allocator<void> > const>)#1}::operator()(std::shared_ptr<sensor_msgs::msg::CameraInfo_<std::allocator<void> > const>) const (this=0x7f2d709630a0, 
[... cut ...]
#28 0x00007f2d9f011620 in rclcpp::executors::SingleThreadedExecutor::spin() () from /opt/ros/humble/lib/librclcpp.so
#29 0x00007f2d9f011835 in rclcpp::spin(std::shared_ptr<rclcpp::node_interfaces::NodeBaseInterface>) () from /opt/ros/humble/lib/librclcpp.so
#30 0x00007f2d9f01192f in rclcpp::spin(std::shared_ptr<rclcpp::Node>) () from /opt/ros/humble/lib/librclcpp.so
#31 0x000055694d75ead7 in main (argc=10, argv=0x7ffe65d3ccc8) at /home/ferdinand/my_workspace/src/fcbox/fcbox_drivers/camera/astra_camera/src/main.cpp:20

I could also provide the source code to start camera.

fujitatomoya, 2023-09-09:
`Publisher` object is NULL, so that 1st memory access on `intra_process_is_enabled_` generates SIGSEGV. can you check if `Publisher` object is valid in `astra_camera::PointCloudXyzNode::depthCb`?

```bash
#0 0x00007f2d9eb173d0 in rclcpp::Publisher<std_msgs::msg::String_<std::allocator >, std::allocator >::publish<std_msgs::msg::String_<std::allocator > > (this=0x0, msg=...)
at /opt/ros/humble/include/rclcpp/rclcpp/publisher.hpp:295
#1 0x00007f2d9eb0c072 in astra_camera::PointCloudXyzNode::depthCb (this=0x7f2d70962960,
depth_msg=std::shared_ptr<const sensor_msgs::msg::Image_<std::allocator >> (use count 3, weak count 0) = {...},
info_msg=std::shared_ptr<const sensor_msgs::msg::CameraInfo_<std::allocator >> (use count 9, weak count 0) = {...})
at /home/ferdinand/my_workspace/src/fcbox/fcbox_drivers/camera/astra_camera/src/point_cloud_proc/point_cloud_xyz.cpp:185
...
```

according to the stack trace, i do not see ROS 2 system is responsible to the exception (segmentation fault).

ferdinandfan, 2023-09-27:
thank you so much. The problem is solved by your advice.

fujitatomoya, 2023-09-27:
@ferdinandfan glad to hear you fixed the problem! can you close this issue?

closed 2023-09-27 by clalancette