ros2/ros2 #1483: Unable to Find /set_model_state and /SetEntityState Services to Reset Robot in ROS2 Foxy and Gazebo 11

Mohammed-akhrif, 2023-09-17:
## Bug report

**Required Info:**

- Operating System:
  -Ubuntu 20.04 (replace with your actual OS)
- Installation type:
  - Binaries
- Version or commit hash:
  - ROS2 Foxy Fitzroy
- DDS implementation:
  - Fast-RTPS (or whichever you are using)
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue
Setup ROS2 Foxy and Gazebo 11.
Launch the simulation using my custom robot URDF and launch files.
Run ros2 service list to list available services.
# Command to list services
`ros2 service list`

#### Expected behavior
I expect to see /set_model_state and /SetEntityState services listed so that I can reset my robot's position during simulation., Now I don't have any way to reset my robot position
#### Actual behavior
Neither /set_model_state nor /SetEntityState services are available in the listed services. This prevents me from resetting the robot's position in the simulation.
#### Additional information
I've included Gazebo plugins in my URDF and my other topics and services seem to be functioning well. 
this is myURDF robot core : 
```
<?xml version="1.0"?>

<robot xmlns:xacro="http://www.ros.org/wiki/xacro">

    <xacro:include filename="inertial_macros.xacro" />
    <xacro:include filename="face.xacro" />
   

    <xacro:property name="chassis_length" value="0.335"/>
    <xacro:property name="chassis_width" value="0.265"/>
    <xacro:property name="chassis_height" value="0.138"/>
    <xacro:property name="chassis_mass" value="1.0"/>
    <xacro:property name="wheel_radius" value="0.034"/>
    <xacro:property name="wheel_thickness" value="0.026"/>
    <xacro:property name="wheel_mass" value="0.05"/>
    <xacro:property name="wheel_offset_x" value="0.226"/>
    <xacro:property name="wheel_offset_y" value="0.166"/>
    <xacro:property name="wheel_offset_z" value="0.01"/>
    <xacro:property name="caster_wheel_radius" value="0.01"/>
    <xacro:property name="caster_wheel_mass" value="0.01"/>
    <xacro:property name="caster_wheel_offset_x" value="0.075"/>
    <xacro:property name="caster_wheel_offset_z" value="${wheel_offset_z - wheel_radius + caster_wheel_radius}"/>

    <material name="white">
       <color rgba="1 1 1 1"/>
    </material>

    <material name="orange">
       <color rgba="1 0.3 0.1 1"/>
    </material>

    <material name="blue">
       <color rgba="0.2 0.2 1 1"/>
    </material>

    <material name="black">
       <color rgba="0 0 0 1"/>
    </material>

    <link name="base_link">

    </link>

    <joint name="chassis_joint" type="fixed">
       <parent link="base_link"/>
       <child link="chassis"/>
       <origin xyz="${-wheel_offset_x} 0 ${-wheel_offset_z}"/>
    </joint>

    <link name="chassis">
        <visual>
            <origin xyz="${chassis_length/2} 0 ${chassis_height/2}"/>
            <geometry>
                <box size="${chassis_length} ${chassis_width} ${chassis_height}"/>
            </geometry>
            <material name="orange"/>
        </visual>
        <collision>
            <origin xyz="${chassis_length/2} 0 ${chassis_height/2}"/>
            <geometry>
               <box size="${chassis_length} ${chassis_width} ${chassis_height}"/>
            </geometry>
        </collision>
        <xacro:inertial_box mass="0.5" x="${chassis_length}" y="${chassis_width}" z="${chassis_height}">
            <origin xyz="${chassis_length/2} 0 ${chassis_height/2}" rpy="0 0 0"/>
        </xacro:inertial_box>

    </link>

    <gazebo reference="chassis">
        <material>Gazebo/Yellow</material>
    </gazebo>

    <!-- LEFT WHEEL -->

    <joint name="left_wheel_joint" type="continuous">
        <parent link="base_link"/>
        <child link="left_wheel"/>
        <origin xyz="0 ${wheel_offset_y} 0" rpy="-${pi/2} 0 0" />
        <axis xyz="0 0 1"/>
    </joint>

    <link name="left_wheel">
        <visual>
            <geometry>
                <cylinder radius="${wheel_radius}" length="${wheel_thickness}"/>
            </geometry>
            <material name="blue"/>
        </visual>
        <collision>
            <geometry>
               <sphere radius="${wheel_radius}"/>
            </geometry> 
        </collision>
        <xacro:inertial_cylinder mass="${wheel_mass}" length="${wheel_thickness}" radius="${wheel_radius}">
            <origin xyz="0 0 0" rpy="0 0 0"/>
        </xacro:inertial_cylinder>
    </link>

    <gazebo reference="left_wheel">
        <material>Gazebo/Blue</material>
    </gazebo>

    <!-- RIGHT WHEEL -->

    <joint name="right_wheel_joint" type="continuous">
        <parent link="base_link"/>
        <child link="right_wheel"/>
        <origin xyz="0 ${-wheel_offset_y} 0" rpy="${pi/2} 0 0" />
        <axis xyz="0 0 -1"/>
    </joint>

    <link name="right_wheel">
        <visual>
            <geometry>
                <cylinder radius="${wheel_radius}" length="${wheel_thickness}"/>
            </geometry>
            <material name="blue"/>
        </visual>
        <collision>
            <geometry>
                <sphere radius="${wheel_radius}"/>
            </geometry>
        </collision>
        <xacro:inertial_cylinder mass="${wheel_mass}" length="${wheel_thickness}" radius="${wheel_radius}">
            <origin xyz="0 0 0" rpy="0 0 0"/>
        </xacro:inertial_cylinder>
    </link>

    <gazebo reference="right_wheel">
        <material>Gazebo/Blue</material>
    </gazebo>

    <!-- Caster WHEEL -->
    <joint name="caster_wheel_joint" type="fixed">
        <parent link="chassis"/>
        <child link="caster_wheel"/>
        <origin xyz="${caster_wheel_offset_x} 0 ${caster_wheel_offset_z}"/>
    </joint>

    <link name="caster_wheel">
        <visual>
            <geometry>
               <sphere radius="${caster_wheel_radius}"/>
            </geometry>
            <material name="white"/>
        </visual>
        <collision>
            <geometry>
                <sphere radius="${caster_wheel_radius}"/>
            </geometry>
        </collision>
        <xacro:inertial_sphere mass="0.1" radius="0.05">
            <origin xyz="0 0 0" rpy="0 0 0"/>
        </xacro:inertial_sphere>
    </link>

    <gazebo reference="caster_wheel">
        <material>Gazebo/White</material>
        <mu1 value="0.001"/>
        <mu2 value="0.001"/>   
    </gazebo>
</robot>

```
and this my robot controller : 
```
<?xml version="1.0"?>
<robot xmlns:xacro="http://www.ros.org/wiki/xacro">

    <ros2_control name="GazeboSystem" type="system">
        <hardware>
            <plugin>gazebo_ros2_control/GazeboSystem</plugin>
        </hardware>
        <joint name="left_wheel_joint">
            <command_interface name="velocity">
                <param name="min">-100</param>
                <param name="max">100</param>
            </command_interface>
            <state_interface name="velocity"/>
            <state_interface name="position"/>
        </joint>
        <joint name="right_wheel_joint">
            <command_interface name="velocity">
                <param name="min">-100</param>
                <param name="max">100</param>
            </command_interface>
            <state_interface name="velocity"/>
            <state_interface name="position"/>
        </joint>
    </ros2_control>

    <gazebo>
        <plugin name="gazebo_ros2_control" filename="libgazebo_ros2_control.so">
            <parameters>$(find my_bot)/config/my_controllers.yaml</parameters>
        </plugin>     
    </gazebo>
</robot>

```

hbonnavaud, 2024-04-15:
Hi, have you found a solution ? I have the same issue

RiccardoMengozzi, 2025-03-13:
I had the same problem, I am now able to see and use the `/set_entity_state` service. All I did was adding this line:
```
<plugin name='gazebo_ros_state' filename='libgazebo_ros_state.so'/>
```
At the top of the `.world` file:
```
<?xml version="1.0"?>
<sdf version="1.6">
  <world name="default">
    <plugin name='gazebo_ros_state' filename='libgazebo_ros_state.so'/>
```

closed 2025-03-20 by mjcarroll