ros2/ros2 #1487: rclcpp::Node::get_topic_names_and_types() can't get active topics

doupongzeng, 2023-09-26:
## Bug report

**Required Info:**

- Operating System: 
  - Ubuntu 20.04
- Installation type: 
  - Binary
- Version or commit hash: 
  - foxy
- DDS implementation: 
  - fast-DDS
- Client library (if applicable): 
  - rclcpp

#### Steps to reproduce issue

```cpp
  rclcpp::Rate r(100);
  int i = 0;
  while (rclcpp::ok() && i < 100) {
    rclcpp::spin_some(node_);
    r.sleep();
    i++;
  }

  auto topics_name_type_ = node_->get_topic_names_and_types();

  for (const auto& topic_it : topics_name_type_) {
    std::string topic_name = topic_it.first;
    std::vector<std::string> topic_types = topic_it.second;
    // iterate over all topic types
    for (const auto& topic_type : topic_types)
      RCLCPP_INFO(node_->get_logger(), "Topic: %s, type: %s", topic_name.c_str(), topic_type.c_str());
  }

```

#### Expected behavior
list all topics like `ros2 topic list`

#### Actual behavior
only list topic 
```
[INFO] [1695696849.853925012] [gui_image]: Topic: /parameter_events, type: rcl_interfaces/msg/ParameterEvent
[INFO] [1695696849.853985715] [gui_image]: Topic: /rosout, type: rcl_interfaces/msg/Log
```

#### Additional information

#### related issue
[issue-739857447](https://github.com/ros2/ros2/issues/1057#issue-739857447)

clalancette, 2023-09-26:
Sorry, Foxy is End-of-Life now.  If you can retry on Humble or Iron, that would help here.  Also, I'll suggest opening questions like this on https://robotics.stackexchange.com/questions/tagged/ros, which is our central Question and Answer site.  You'll get a better answer there, and it will be searchable for the future.

closed 2023-09-26 by clalancette

doupongzeng, 2023-09-28:
> Sorry, Foxy is End-of-Life now. If you can retry on Humble or Iron, that would help here. Also, I'll suggest opening questions like this on https://robotics.stackexchange.com/questions/tagged/ros, [...]

OK, This community forum looks good.