ros2/ros2 #1488: Multiple TSAN report when running the system

Rrooach, 2023-09-27:
## Bug report

**Required Info:**

- Operating System:
  Ubuntu 22.04
- Installation type:
  source code
- Version or commit hash:
  humble
- DDS implementation:
  FastRTPS
- Client library (if applicable):
  RCLCPP

#### Steps to reproduce issue
Basically, I installed ros-humble on a remote server, compiled it with ThreadSanitizer, And when I start to boot the ros2 system and try to testing it, I get multiple TSAN report.

#### Expected behavior
The system start without error.

#### Actual behavior
System emit 2 different TSAN error report, within 1 in subscription.c (possible deadlock), 1 in FlowControllerImpl.hpp (data race).

#### Additional information
__1. data race in FlowControllerImpl.hpp:1186 and FlowControllerImpl.hpp:147__

report link: https://pastebin.com/FdGR9iC1

It seems there are a race condition between 
```CPP
        if (nullptr == change->writer_info.previous &&
                nullptr == change->writer_info.next)
        {
            std::unique_lock<std::mutex> lock(async_mode.changes_interested_mutex);
            sched.add_old_sample(writer, change);
            async_mode.cv.notify_one();

            return true;
        }

```
and 
```CPP
      void add_list(
                ListInfo& list) noexcept
        {
            if (!list.is_empty())
            {
                fastrtps::rtps::CacheChange_t* first = list.head.writer_info.next;
                fastrtps::rtps::CacheChange_t* last = list.tail.writer_info.previous;

                first->writer_info.previous = tail.writer_info.previous;
                first->writer_info.previous->writer_info.next = first;
                last->writer_info.next = &tail;
                tail.writer_info.previous = last;

                list.clear();
            }

```

__2. lock-order-inversion in subscription.c:218__

report link: https://pastebin.com/958rTvtB

when rmw is tring to destory the subscription
```cpp
rmw_ret_t
destroy_subscription(
  const char * identifier,
  CustomParticipantInfo * participant_info,
  rmw_subscription_t * subscription,
  bool reset_cft)
{
  assert(subscription->implementation_identifier == identifier);
  static_cast<void>(identifier);

  {
    std::lock_guard<std::mutex> lck(participant_info->entity_creation_mutex_);

    // Get RMW Subscriber
    auto info = static_cast<CustomSubscriberInfo *>(subscription->data);

```

Another thread is still holding this sub
```CPP
    std::lock_guard<std::mutex> guard(common_context->node_update_mutex);
    rmw_gid_t request_publisher_gid = rmw_fastrtps_shared_cpp::create_rmw_gid(
      eprosima_fastrtps_identifier, info->request_writer_->guid());
``` 

It would be so nice of you to review this bug report. Thank you again for taking the time. Any suggestions or feedback you can provide would be very helpful.

fujitatomoya, 2023-09-27:
@Rrooach thanks for posting issue.

can you share the whole procedure including building, test samples and exec command lines?

Rrooach, 2023-09-28:
> @Rrooach thanks for posting issue.
> 
> can you share the whole procedure including building, test samples and exec command lines?
Basically, I build the ros2 system using tsan, 
```shell
  colcon build --parallel-workers 32. --cmake-args -DCMAKE_CXX_FLAGS="-Wno-error -g -fsanitize=thread " -DCMAKE_C_FLAGS="-Wno-error -g -fsanitize=thread "
```
And we boot the ros2 system with our fuzzer, and get the error report,
However, we find that to trigger such issues, we can also run test_graph__rmw_fastrtps_cpp within the build/rcl_action

fujitatomoya, 2023-09-28:
CC: @iuhilnehc-ynos @Barry-Xu-2018

iuhilnehc-ynos, 2023-10-09:
Let's focus on these two TSAN warnings first.

- data race

I think the https://pastebin.com/FdGR9iC1 is a data race even though there are [comments](https://github.com/eProsima/Fast-DDS/blob/7f7c90994ad83c169b52c397fbdbac4525a7728f/src/cpp/rtps/flowcontrol/FlowControllerImpl.hpp#L1181-L1185) to explain that it's a thread-safe for the comparison.

From the report, accessing the `change->writer_info.previous or change->writer_info.next` in T16(mutexes: write M18210) and T37(mutexes: write M18120, write M18121) is not protected by one same mutex.

@MiguelCompany Do you have any suggestion?

- lock-order-inversion

It's a potential dead-lock.

　　[rmw_create_client](https://github.com/ros2/rmw_fastrtps/blob/fe73970c674af7ff286e5a77337dc11865fc2691/rmw_fastrtps_cpp/src/rmw_client.cpp#L57)
　　　　[lock A with participant_info->entity_creation_mutex_](https://github.com/ros2/rmw_fastrtps/blob/fe73970c674af7ff286e5a77337dc11865fc2691/rmw_fastrtps_cpp/src/rmw_client.cpp#L130)
　　　　[lock B with common_context->node_update_mutex](https://github.com/ros2/rmw_fastrtps/blob/fe73970c674af7ff286e5a77337dc11865fc2691/rmw_fastrtps_cpp/src/rmw_client.cpp#L440)
		
　　[rmw_create_subscription](https://github.com/ros2/rmw_fastrtps/blob/79f23ae421d2ad21e8e191f5adb8d104dde2fe4b/rmw_fastrtps_cpp/src/rmw_subscription.cpp#L58)
　　　　[lock B with common_context->node_update_mutex](https://github.com/ros2/rmw_fastrtps/blob/79f23ae421d2ad21e8e191f5adb8d104dde2fe4b/rmw_fastrtps_cpp/src/rmw_subscription.cpp#L90)
　　　　[rmw_fastrtps_shared_cpp::destroy_subscription](https://github.com/ros2/rmw_fastrtps/blob/79f23ae421d2ad21e8e191f5adb8d104dde2fe4b/rmw_fastrtps_cpp/src/rmw_subscription.cpp#L104)
　　　　　　[lock A with participant_info->entity_creation_mutex_](https://github.com/ros2/rmw_fastrtps/blob/490f06edd5c0cc7424d485ac9ea85e0a24158b99/rmw_fastrtps_shared_cpp/src/subscription.cpp#L44)

@fujitatomoya Any suggestions to fix it?

NOTE: Besides the above two issues, there are another TSAN warnings(not checked yet).

fujitatomoya, 2023-10-09:
> It's a potential dead-lock.

i think that `publisher` or `subscription` for creation or destruction does not have this problem.

> [rmw_create_client](https://github.com/ros2/rmw_fastrtps/blob/fe73970c674af7ff286e5a77337dc11865fc2691/rmw_fastrtps_cpp/src/rmw_client.cpp#L57)
> 　　　　[lock A with [...]

the main difference is, `participant_info->entity_creation_mutex_` is acquired only for dds entities creations, and will be released to node graph change. probably we can align with those implementation for `client` and `service` as well? if we do that, i think this specific inversion will be avoided.

fujitatomoya, 2023-10-30:
Related PRs:

- https://github.com/ros2/rmw_dds_common/pull/73 (DDS generic change)
- https://github.com/ros2/rmw_fastrtps/pull/725
- https://github.com/ros2/rmw_cyclonedds/pull/474
- https://github.com/ros2/rmw_connextdds/pull/134

clalancette, 2023-11-03:
All right, all linked PRs are in, so closing this out.

closed 2023-11-03 by clalancette