ros2/ros2 #1489: [humble]strange bug on windows11

sdlizheng2015, 2023-09-27:
## Bug report

Hello, everyone, I am using ros2 humble on windows11. Today I write my code and compile and run them, all of them is normal. However, in the afternoon some bug happens.

I can not use "ros2 run" or "ros2 launch" to run my packages, including ros2 demo(turtlesim and so on). It is very strange, because I can still use "ros2 create" and can compile my code.

When I use "ros2 run" or "ros2 launch", the cmd has nothing to print or display so that I can not get any information about error. It seems like the program is in stuck.

Besides, I also can not use rqt. I can open it, but if I want to publish topics, it will be stuck.

Looking forward to your reply，thanks！

**Machine Info:**

- Operating System: Windows 11
- Installation type: binaries 
- Version or commit hash: humble
- DDS implementation: Fast-RTPS
- Client library (if applicable): rclcpp

mjcarroll, 2023-10-13:
> I can not use "ros2 run" or "ros2 launch" to run my packages, including ros2 demo(demo_nodes_cpp, turtlesim and so on). It is very strange, because I can still use "ros2 create" and can compile my code.

Can you directly run the nodes in question? For example `./lib/demo_nodes_cpp/demo_nodes_cpp.exe`

sdlizheng2015, 2023-10-14:
> Can you directly run the nodes in question? For example `./lib/demo_nodes_cpp/demo_nodes_cpp.exe`

Unlucky, can not run these nodes

sdlizheng2015, 2023-12-05:
> You can try this method: Deleting C:\ProgramData\eprosima\fastrtps_interprocess and rebooting.

Thanks! I will try.

chris2lee, 2024-07-23:
> You can try this method: Deleting C:\ProgramData\eprosima\fastrtps_interprocess and rebooting.

It's useful, thanks!

grahas, 2024-12-06:
> You can try this method: Deleting C:\ProgramData\eprosima\fastrtps_interprocess and rebooting.

@q2013 I tried this and it worked! Do you or anyone have any explanation as to what is happening and why this fixes it?

mjcarroll, 2025-03-29:
Looks like there was some amount of resolution here.

Closing as stale, feel free to re-open if you have further information or questions.

closed 2025-03-29 by mjcarroll