ros2/ros2 #1491: Installation issues when installing Iron from source on Fedora 39

toitimtoe, 2023-10-08:
## Bug report and proposed ways to help in the documentation

**Required Info:**

- Operating System:
  - Fedora 39
- Installation type:
  - From source according to [this documentation](https://docs.ros.org/en/iron/Installation/Alternatives/RHEL-Development-Setup.html#build-the-code-in-the-workspace)
- Version or commit hash:
  - [repos file from the documentation](https://raw.githubusercontent.com/ros2/ros2/iron/ros2.repos)
- DDS implementation:
  - Default Fast DDS
- Client library (if applicable):
  - Unsure

#### Steps to reproduce issue
Followed the instructions step by step from [this documentation](https://docs.ros.org/en/iron/Installation/Alternatives/RHEL-Development-Setup.html#build-the-code-in-the-workspace).

#### Expected behavior
Installed ros2 Iron without issues

#### Actual behavior
Failed to build some packages when running `colcon build --symlink-install`

#### Additional information
Two packages that failed:
##### `rcpputils` 
The file that caused the issue is `src/ros2/rcpputils/include/rcpputils/filesystem_helper.hpp`.

```shell
filesystem_helper.hpp:137:20: error: ‘uint64_t’ does not name a type   137 |   RCPPUTILS_PUBLIC uint64_t file_size() const;
```

I found that Chris fixed this issue in a [PR](https://github.com/ros2/rcpputils/pull/178) a while back, but it's probably not updated to the repo files.
My solution is to apply the fix on the file locally. We should update the repo files so that other users do not run into this same issue.

##### `qt_gui_cpp`
 
Errors are such as 

```shell
g++: fatal error: environment variable ‘RPM_ARCH’ not defined
```

and 

```shell
g++: fatal error: environment variable ‘RPM_PACKAGE_RELEASE’ not defined
```

Solution (inspired by [the oolite installation troubleshooting](https://github.com/OoliteProject/oolite/blob/master/README.md#troubleshooting)):

```shell
export RPM_ARCH=$(uname -m)
export RPM_PACKAGE_RELEASE=1.0
export RPM_PACKAGE_VERSION=1.0
export RPM_PACKAGE_VERSION=qt_gui_cpp
```

I filled random values for the `RPM_PACKAGE_RELEASE`, `RPM_PACKAGE_VERSION` and `RPM_PACKAGE_NAME`. Not sure what are the correct values, but I don't think they would affect the installation process.

Hope this helps any fellow fedora users out there. Do let me know if there is a better way of doing this or documenting this.

clalancette, 2023-10-09:
It was just never backported to Iron.  I've now done that in https://github.com/ros2/rcpputils/pull/182

toitimtoe, 2023-10-09:
> It was just never backported to Iron. I've now done that in [ros2/rcpputils#182](https://github.com/ros2/rcpputils/pull/182)

Thank you so much for the quick response.
Can we also backport the similar fixes in [rclcpp](https://github.com/ros2/rclcpp/pull/2186/) and [libstatistic-collector](https://github.com/ros-tooling/libstatistics_collector/pull/165) please?

closed 2023-11-18 by toitimtoe

EnteRente, 2024-06-27:
> ## Bug report and proposed ways to help in the documentation
> 
> **Required Info:**
> 
>     * Operating System:
>       
>       * Fedora 39
> 
>     * Installation type:
>       
>       * From source according to [this documentation](https://docs.ros.org/en/iron/Installation/Alternatives/RHEL-Development-Setup.html#build-the-code-in-the-workspace)
> 
>     * Version or commit hash:
>       
>       * [repos file from the documentation](https://raw.githubusercontent.com/ros2/ros2/iron/ros2.repos)
> 
>     * DDS implementation:
>       
>       * Default Fast DDS
> 
>     * Client library (if applicable):
>       
>       * Unsure
> 
> 
> #### Steps to reproduce issue
> 
> Followed the instructions step by step from [this documentation](https://docs.ros.org/en/iron/Installation/Alternatives/RHEL-Development-Setup.html#build-the-code-in-the-workspace).
> #### Expected behavior
> 
> Installed ros2 Iron without issues
> #### Actual behavior
> 
> Failed to build some packages when running `colcon build --symlink-install`
> #### Additional information
> ##### `rcpputils`
> 
> The file that caused the issue is `src/ros2/rcpputils/include/rcpputils/filesystem_helper.hpp`.
> 
> ```shell
> filesystem_helper.hpp:137:20: error: ‘uint64_t’ does not name a type   137 |   RCPPUTILS_PUBLIC uint64_t file_size() const;
> ```
> 
> I found that Chris fixed this issue in a [PR](https://github.com/ros2/rcpputils/pull/178) a while back, but it's probably not updated to the repo files. My workaround is to apply the fix on the file locally.
> 
> This issue seems to also affect `ros-tooling/libstatistics_collector/include/libstatistics_collector/moving_average_statistics/types.hpp`, fixed in [PR](https://github.com/ros-tooling/libstatistics_collector/pull/165).
> 
> A missing `stdexcept` for `rclcpp` is also fixed in [PR](https://github.com/ros2/rclcpp/pull/2186) but not merged into Iron.
> ##### `qt_gui_cpp`
> 
> Errors are such as
> 
> ```shell
> g++: fatal error: environment variable ‘RPM_ARCH’ not defined
> ```
> 
> and
> 
> ```shell
> g++: fatal error: environment variable ‘RPM_PACKAGE_RELEASE’ not defined
> ```
> 
> Solution (inspired by [the oolite installation troubleshooting](https://github.com/OoliteProject/oolite/blob/master/README.md#troubleshooting)):
> 
> ```shell
> export RPM_ARCH=$(uname -m)
> export RPM_PACKAGE_RELEASE=1.0
> export RPM_PACKAGE_VERSION=1.0
> export RPM_PACKAGE_VERSION=qt_gui_cpp
> ```
> 
> I filled random values for the `RPM_PACKAGE_RELEASE`, `RPM_PACKAGE_VERSION` and `RPM_PACKAGE_NAME`. Not sure what are the correct values, but I don't think they would affect the installation process.
> 
> Hope this helps any fellow fedora users out there. Do let me know if there is a better way of doing this or documenting this.

Thank you, this helped a lot. There is just a minor typo. Nothing significant, but I just wanted to point it out. It should be `export RPM_PACKAGE_NAME=qt_gui_cpp` instead of `export RPM_PACKAGE_VERSION=qt_gui_cpp`