ros2/ros2 #1494: [RTPS_PARTICIPANT Error] Could not create the specified receiver resource

fujitatomoya, 2023-10-24:
## Bug report

**Required Info:**

- Operating System:
  - ubuntu 22.04
- Installation type:
  - source build
- Version or commit hash:
  - https://github.com/ros2/ros2/commit/1f5bd8ed43beea199dabe48bc8023af3aba9806c
- DDS implementation:
  - Fast-DDS
- Client library (if applicable):
  - rclpy (ros2cli)

#### Steps to reproduce issue

- terminal-1

```bash
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 run demo_nodes_cpp parameter_blackboard
[INFO] [1698124475.161121870] [parameter_blackboard]: Parameter blackboard node named '/parameter_blackboard' ready, and serving '6' parameters already!
```

- terminal-2

```bash
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 param list
2023-10-23 22:14:38.938 [RTPS_PARTICIPANT Error] Could not create the specified receiver resource -> Function createReceiverResources
2023-10-23 22:14:38.938 [RTPS_PARTICIPANT Error] Could not create the specified receiver resource -> Function createReceiverResources
2023-10-23 22:14:38.938 [RTPS_PARTICIPANT Error] Could not create the specified receiver resource -> Function createReceiverResources
2023-10-23 22:14:39.478 [RTPS_PARTICIPANT Error] Could not create the specified receiver resource -> Function createReceiverResources
2023-10-23 22:14:39.479 [RTPS_PARTICIPANT Error] Could not create the specified receiver resource -> Function createReceiverResources
2023-10-23 22:14:39.539 [RTPS_PARTICIPANT Error] Could not create the specified receiver resource -> Function createReceiverResources
/parameter_blackboard:
  qos_overrides./parameter_events.publisher.depth
  qos_overrides./parameter_events.publisher.durability
  qos_overrides./parameter_events.publisher.history
  qos_overrides./parameter_events.publisher.reliability
  start_type_description_service
  use_sim_time
```

#### Expected behavior

Those error should not print.

```bash
2023-10-23 22:14:38.938 [RTPS_PARTICIPANT Error] Could not create the specified receiver resource -> Function createReceiverResources
```

#### Actual behavior

The following errors print every time using `ros2cli`.

```bash
2023-10-23 22:14:38.938 [RTPS_PARTICIPANT Error] Could not create the specified receiver resource -> Function createReceiverResources
```

fujitatomoya, 2023-10-24:
CC: @iuhilnehc-ynos @Barry-Xu-2018 @MiguelCompany

iuhilnehc-ynos, 2023-10-24:
It's a known issue. https://github.com/eProsima/Fast-DDS/pull/3936#issuecomment-1768541925
I think it will be addressed by this way. https://github.com/eProsima/Fast-DDS/pull/3938#issuecomment-1772162216.

fujitatomoya, 2023-10-24:
@iuhilnehc-ynos thanks! i will keep eyes on https://github.com/eProsima/Fast-DDS/pull/3936#issuecomment-1768541925.

MiguelCompany, 2023-10-24:
@fujitatomoya See eProsima/Fast-DDS#3955

fujitatomoya, 2023-10-24:
confirmed https://github.com/eProsima/Fast-DDS/pull/3955 can fix this problem with `rolling`.

fujitatomoya, 2023-10-24:
I will go head to close this and we can keep tracking on backporting PRs via https://github.com/ros2/ros2cli/issues/861

closed 2023-10-24 by fujitatomoya