ros2/ros2 #1500: ROS2 listener/talker only works for talker on MacOS Sonoma 14.2 on M1-Max within docker

nihavend, 2023-12-24:
## Bug report

**Required Info:**

- Operating System:
  - MacOS Sonoma 14.2 
- Installation type:
  - docker
- Version or commit hash:
  - osrf/ros:iron-desktop 2b4beafb3bbc
- DDS implementation:
  - 
- Client library (if applicable):
  - 

#### Steps to reproduce issue

```
ros2 run demo_nodes_cpp listener &
ros2 run demo_nodes_cpp talker
```

#### Expected behavior
talker/listener should work as defined in documentation
#### Actual behavior
talker/listener does not work as defined in documentation and needs **_--net=host_** parameter while starting docker as defined in [here](https://robotics.stackexchange.com/questions/107065/ros2-listener-talker-only-works-for-talker-on-macos-sonoma-14-2-on-m1-max-within)

Bu it is a bit wierd because both talker/listener are within same container and should work without need of host networking
#### Additional information
folowed steps from [here](https://docs.ros.org/en/iron/How-To-Guides/Run-2-nodes-in-single-or-separate-docker-containers.html) for single docker case

----
## Feature request

#### Feature description

#### Implementation considerations

clalancette, 2023-12-26:
macOS is a community-supported platform for ROS 2, so we don't do regular testing on it.  If you find out how to fix the problem, please feel free to open PRs to update either the code or the documentation.

fujitatomoya, 2023-12-26:
@nihavend i cannot reproduce this issue with linux platform (ubuntu 22.04), i would check the network interfaces assigned to the container, there should be network interface provided by docker bridge and loopback. so that DDS (Fast-DDS) can see both endpoints are in the same local host then use [shared memory transport](https://fast-dds.docs.eprosima.com/en/latest/fastdds/transport/shared_memory/shared_memory.html#shared-memory-transport) (`/dev/shm`) for the communication.

nihavend, 2023-12-27:
@clalancette i am not sure about the root cause of the problem.  @fujitatomoya I confirm it does not happen on Ubuntu, it works without `--net=host` parameter. I do not think it is acceptable to make it work over host even both talker/listener in same container. Need some investigation which is beyond my knowledge on ROS2 code base even for today