ros2/ros2 #1501: A new version startup reported a Python error

Jzow, 2023-12-26:
When I started ROS2, the following error occurred. I was running on Windows 10

Python:3.8.3
ROS2:humble

PKG_CONFIG_PATH=D:\ros2\lib\pkgconfig
PYTHONPATH=C:\ci\ws\install\lib\python3.8\dist-packages;D:\ros2\Lib\site-packages
ROS_DISTRO=humble
ROS_LOCALHOST_ONLY=1
ROS_PYTHON_VERSION=3
ROS_VERSION=2

```python
C:\Users\Administrator>ros2 pkg executables turtlesim
Traceback (most recent call last):
  File "D:\ros2\Scripts\ros2-script.py", line 33, in <module>
    sys.exit(load_entry_point('ros2cli==0.18.7', 'console_scripts', 'ros2')())
  File "D:\ros2\Scripts\ros2-script.py", line 25, in importlib_load_entry_point
    return next(matches).load()
  File "C:\Python38\lib\importlib\metadata.py", line 77, in load
    module = import_module(match.group('module'))
  File "C:\Python38\lib\importlib\__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 783, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "D:\ros2\Lib\site-packages\ros2cli\cli.py", line 22, in <module>
    from ros2cli.command import add_subparsers_on_demand
  File "D:\ros2\Lib\site-packages\ros2cli\command\__init__.py", line 23, in <module>
    from ros2cli.plugin_system import instantiate_extensions
  File "D:\ros2\Lib\site-packages\ros2cli\plugin_system.py", line 19, in <module>
    from packaging.version import Version
ModuleNotFoundError: No module named 'packaging'
```

No matter what cli command is executed, he will report this error

Jzow, 2023-12-26:
My Python environment is located in `C:\Python38`, and I would like to know if it is necessary to modify the configuration of ROS2. Python `PYTHONPATH=C:\ci\ws\install\lib\python3.8\dist-packages;D:\ros2\Lib\site-packages` to `C:\Python38`

clalancette, 2024-01-04:
It shouldn't be necessary to modify the configuration, no.

What it looks like here is that you are missing the `packaging` pip package.  Can you run `pip freeze --all` command and paste the results?

Jzow, 2024-01-05:
I try, thanks

clalancette, 2024-09-10:
There's been no update to this for many months, so I'm going to close this for now.  If you are still having problems, please feel free to reopen and provide the requested information.

closed 2024-09-10 by clalancette