ros2/ros2 #1505: file not exits: /opt/ros/foxy/setup.bash

Chortine, 2024-01-04:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - Ubuntu (Debian) https://docs.ros.org/en/foxy/Installation/Ubuntu-Install-Debians.html#ubuntu-debian
- Version or commit hash:
  - ros foxy

#### Bug description
I install the ros foxy via ubuntu debian, after I follow all the steps listed in https://docs.ros.org/en/foxy/Installation/Ubuntu-Install-Debians.html#ubuntu-debian, the apt shows successful installation. However, in /opt/ros/foxy directory, I found all the sub-directories, but no the  setup.bash files that are supposed to be there.
[image]

fujitatomoya, 2024-01-04:
@Chortine thanks for creating issue.

unfortunately foxy is already End Of Life, https://docs.ros.org/en/rolling/Releases.html. can you try `Humble` or later, and if you see the same problem please open the another issue for that.

fujitatomoya, 2024-01-04:
@clalancette can you close this?

Chortine, 2024-01-04:
> @Chortine thanks for creating issue.
> 
> unfortunately foxy is already End Of Life, https://docs.ros.org/en/rolling/Releases.html. can you try `Humble` or later, and if you see the same problem please open the another issue for that.

sorry I'm still use ros foxy because I'm using ubuntu 20.04. As far as I know ros humble only works for 22.04, right? Updating to 22.04 will effect lots of dependencies on my env, so I would like to keep current release version.

clalancette, 2024-01-04:
> sorry I'm still use ros foxy because I'm using ubuntu 20.04. As far as I know ros humble only works for 22.04, right?

You can compile Humble from source on Ubuntu 20.04.  I'll suggest doing that, as it will be supported until 2027.

If you can't do that, then I'll suggest opening a ticket on https://robotics.stackexchange.com/questions/tagged/ros, which is our central Question and Answer site.

Make sure to include a lot of information on what platform you are using, which ROS distribution you are using, and the exact steps you took.

closed 2024-01-04 by clalancette