ros2/ros2 #1511: Transition of Rolling to Ubuntu 24.04

nuclearsandwich, 2024-01-16:
Borrowing heavily from https://github.com/ros2/ros2/issues/1221, here's a detailed checklist of items to take care of for the Rolling transition to Ubuntu 24.04.
- [x] Create bootstrap repositories for `noble` and `trixie`
- [ ] Create ROS 2 repositories for `noble` and `trixie`
  - https://github.com/osrf/chef-osrf/pull/228 (private chef link)
- [ ] Create osrf qemu-user-static images https://github.com/osrf/multiarch-docker-image-generation/pull/40
  - [ ] osrf/ubuntu_arm64:noble
  - [ ] osrf/debian_arm64:trixie 
- [ ] Add Ubuntu Noble support to [ros2/ci](https://github.com/ros2/ci) 
  - [ ] Check target Connext version for Rolling and make sure it matches the intended version for Jazzy.
- [ ] Prepare updates to rosdep database to make sure that dependencies are met on Ubuntu Noble
- [ ] Perform a test migration of ROS 2 Rolling to its new target platforms
- [ ] Perform the migration of ROS 2 Rolling to its new target platforms
- [ ] Update target platforms for build farm jobs
- [ ] Update the default platforms for Rolling jobs in [ros2/ci](https://github.com/ros2/ci)

berndpfrommer, 2024-04-04:
Some of my packages that built on the ros2 build farm when Rolling was under Jammy fail to build now under Noble.
Where is the right forum to ask for help regarding this?
Concretely, the problem is that libavcodec is not installed properly:

```
02:12:13 Investigating (0) libavcodec-dev:amd64 < none -> 7:6.1.1-1ubuntu1 @un puN Ib >
02:12:13 Broken libavcodec-dev:amd64 Depends on libavcodec60:amd64 < none @un H > (= 7:6.1.1-1ubuntu1)

```
The full build log is [here](https://build.ros2.org/job/Rbin_uN64__metavision_driver__ubuntu_noble_amd64__binary/31/console)

clalancette, 2024-04-04:
Ubuntu 24.04 is still in the middle of a rebuild due to the xz vulnerability late last week.  So there are still lots of updates trickling down.  I would say wait a few more days; if it still doesn't work by the middle of next week, then we can start taking a closer look.

clalancette, 2024-06-20:
This was done, so closing this out.