ros2/ros2 #1515: ros2 cli hangs on arm in humble

john-maidbot, 2024-01-30:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04, aarch64 (ARM)
- Installation type:
  - binaries (debian apt install)
- DDS implementation:
  - Cyclone DDS
- Client library (if applicable):
  - rclcpp, rclpy

#### Steps to reproduce issue

Run any sort of ros2 command (ros2 topic ..., ros2 service ...) on my ARM system (not seeing it on my x86 system with identical configuration).

#### Expected behavior
The ros2 cli commands return eventually.

#### Actual behavior
The ros2 cli commands just hang. htop suggests it is working on something. But minutes go by and no output is returned.

#### Additional information
This started after I updated my ros2 packages from the recent humble sync. There was conspicuously a ros-humble-cyclonedds update, which makes me suspicious that the rmw could be related. BUT i also tried with RMW_IMPLEMENTATION=rmw_fastrtps_cpp, and that has the same problem.

john-maidbot, 2024-01-30:
ah i think its the daemon thats hung.
if i run ros2 topic echo/list commands with --no-daemon, it works properly.
however trying to stop/start the daemon hangs.

fujitatomoya, 2024-01-31:
@john-maidbot https://github.com/ros2/ros2cli/pull/852 could fix the problem. I am not sure what distro you use, but all backportings are completed. if that fix is not available in release, can you try with source build with your environment?

fujitatomoya, 2024-01-31:
@john-maidbot is this reproducible on ARM platform? any specific procedure to make it happen? for example, do we need remote nodes via network? could you describe the reproducible environment in detail?

john-maidbot, 2024-01-31:
so far i have repeated this on 3 ARM computers (imx8qm SoM's running Ubuntu 22.04 + ROS2 humble in a docker container). 
if i start the container, even without running any nodes, and run ros2 topic list, it has this behavior.
I will try to make a minimal docker container and see if it still happens. if so i will share it here. either way i'll report back

fujitatomoya, 2024-01-31:
so far, i cannot reproduce this problem with Raspi4 Ubuntu20.04 host.

```bash
ubuntu@raspi4-1:~$ for((i=1;i<=100;i+=1)); do docker run --rm -it ros:humble /bin/bash -c "ros2 topic list"; done
/parameter_events
/rosout
/parameter_events
/rosout
/parameter_events
/rosout
/parameter_events
/rosout
/parameter_events
/rosout
/parameter_events
/rosout
/parameter_events
/rosout
...
```

john-maidbot, 2024-01-31:
ah interesting, that also works for me :open_mouth:  

yet in my custom ros2 humble container, it just hangs. :thinking:  i will dig into the diff between ros:humble and my container. 

thank you!

john-maidbot, 2024-02-01:
since this seems to be an issue on my setup and not a ros problem, i will close this.

closed 2024-02-01 by john-maidbot