ros2/ros2 #1517: ros2 param dump spells out # Parameter not set

jackiescanlon, 2024-02-02:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- Installation type:
  - Binaries
- Version or commit hash:
  - ros-iron-rclpy apt package version 4.1.4-1jammy.20231205.160931
- DDS implementation:
  - CycloneDDS
- Client library (if applicable):
  - rclcpy

#### Steps to reproduce issue
1. Create a node with this code:
```py
#!/usr/bin/env python3
import rclpy
import rclpy.node
from rcl_interfaces.msg import ParameterDescriptor, ParameterType

class ParamDemo(rclpy.node.Node):
    def __init__(self):
        super().__init__("param_demo")

        self.declare_parameter(
            "param_1",
            "param_1_default_value",
            descriptor=ParameterDescriptor(type=ParameterType.PARAMETER_STRING),
        )

        self.declare_parameter(
            "param_2",
            descriptor=ParameterDescriptor(type=ParameterType.PARAMETER_STRING),
        )

def main():
    rclpy.init()
    node = ParamDemo()
    rclpy.spin(node)

if __name__ == "__main__":
    main()
```
2. Run it:
```
ros2 run my_package parameter_node.py
```

#### Expected behavior
`ros2 param dump /param_demo` shows that `param_1` is set to `param_1_default_value` and `param_2` is unset

#### Actual behavior
`ros2 param dump` reports that `param_1` is set to `#` and `param_2` is set to ` `

#### Additional information

1. If you add several more parameters, you can see the erroneous parameter values spell out `# Parameter not set`, and any parameters that go past that character length (19) are not showed. (You can see `param_9` is omitted in the example below, even though I've declared it).
```
❯ ros2 param dump /param_demo
/param_demo:
  ros__parameters:
    param_1: '#'
    param_10: ' '
    param_11: P
    param_12: a
    param_13: r
    param_14: a
    param_15: m
    param_16: e
    param_17: t
    param_18: e
    param_19: r
    param_2: ' '
    param_20: n
    param_3: o
    param_4: t
    param_5: ' '
    param_6: s
    param_7: e
    param_8: t
```

3. Individual `ros2 param get`s seem to work fine:
```
❯ ros2 param get /param_demo param_1
String value is: param_1_default_value
❯ ros2 param get /param_demo param_2
Parameter not set
```

fujitatomoya, 2024-02-07:
`rolling` can reproduce this issue, https://github.com/ros2/ros2/commit/0f10c2012877aab87ba5761fa409951eaaa4a1db

```bash
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 param dump /param_demo
/param_demo:
  ros__parameters:
    param_1: '#'
    param_2: ' '
    start_type_description_service: P
    use_sim_time: a

root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 param list
/param_demo:
  param_1
  param_2
  start_type_description_service
  use_sim_time
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 param describe /param_demo param_1
Parameter name: param_1
  Type: string
  Constraints:
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 param get /param_demo param_1
String value is: param_1_default_value
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 param get /param_demo param_2
Parameter not set
```

see https://github.com/fujitatomoya/ros2_test_prover/commit/62e14500a25ba978cea760c7c9b27f17172c0403

there is a couple of things we need to address here.

1. `ros2 param dump` needs to handle empty list as exception if happens instead of proceeding zipping results. if empty result returns from parameter service, it will handle `# Parameter not set` like character array, that leads to this weird printing.
2. `rclpy`, i think this should generate `ValueError` exception, since this is not initializing statically typed parameter.

  ```python
          self.declare_parameter(
              "param_2",
              descriptor=ParameterDescriptor(type=ParameterType.PARAMETER_STRING),
          )
  ```

  `rclcpp` generates the exception.

  ```cpp
    rcl_interfaces::msg::ParameterDescriptor descriptor;
    descriptor.type = rclcpp::ParameterType::PARAMETER_STRING;
    node->declare_parameter<std::string>( // rclcpp::exceptions::UninitializedStaticallyTypedParameterException
      "param_2_cpp",
      descriptor);
  ```

  above should work with `dynamic_typing_= True`.

wjwwood, 2024-02-07:
@fujitatomoya This seems like a change that's needed in rclpy and ros2cli, so I think this repo is ok for now, unless you wan to split it up to the two repositories. Honestly, I'd just leave it here for now.

clalancette, 2024-02-16:
We merged in the fixes here, and backported the ones we could.  So closing this out, thanks for the report.

closed 2024-02-16 by clalancette

jackiescanlon, 2024-02-16:
Thank you very much!

jackiescanlon, 2024-04-10:
Following up here, is there any planned date for the next release of `ros2cli` so we can use https://github.com/ros2/ros2cli/pull/881?

Also, I see that https://github.com/ros2/rclpy/pull/1216 is not going to be ported to iron. Is there any alternative fix that could be added for iron?

fujitatomoya, 2024-04-12:
> Following up here, is there any planned date for the next release of ros2cli so we can use https://github.com/ros2/ros2cli/pull/881?

that i do not know, CC: @Yadunund 

> Also, I see that https://github.com/ros2/rclpy/pull/1216 is not going to be ported to iron. Is there any alternative fix that could be added for iron?

im not sure... this is behavior change to generate the exception once it gets merged. that possibly breaks the user space, so we decided not o merge.