ros2/ros2 #1519: colcon build uses wrong version of python if a python version is built from source

geoeo, 2024-02-06:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22
- Installation type:
  - binaries (apt)
- Version or commit hash:
  - Hubmle
- DDS implementation:
  - default eProsima’s Fast DDS.

#### Steps to reproduce issue
- Install python 3.6 from source 
- Install python 3.10 from apt
- install ros2 humble
- build some c++ ros2 package

#### Expected behavior
colcon selects python 3.10

#### Actual behavior
colcon select python3.6

#### Additional information
 Similar to #1421 but no vens are involved
 
 
build output
```

Starting >>> px4_msgs
[Processing: px4_msgs]                             
[Processing: px4_msgs]                                     
[Processing: px4_msgs]                                       
[Processing: px4_msgs]                                       
--- stderr: px4_msgs                                        
/usr/bin/ld: /usr/local/lib/libpython3.6m.a(_sre.o): warning: relocation against `PyExc_RecursionError' in read-only section `.text'
/usr/bin/ld: /usr/local/lib/libpython3.6m.a(bytearrayobject.o): relocation R_X86_64_PC32 against symbol `_Py_NoneStruct' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: final link failed: bad value
collect2: error: ld returned 1 exit status
gmake[2]: *** [CMakeFiles/px4_msgs__rosidl_generator_py.dir/build.make:3386: rosidl_generator_py/px4_msgs/libpx4_msgs__rosidl_generator_py.so] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:475: CMakeFile
```

geoeo, 2024-02-06:
Sure its a path problem, but I removed the offending folders from `PATH` and it still didnt work.  I removed python 3.6 now and I found that rosdep was defaulting to Python.h for python3.6. So maybe rosdep was the offender

Yadunund, 2024-02-07:
`rosdep` is just a python script so it again utilizes whichever python is found first. It might be worth investing in a clean terminal environment, what the outputs of `which python` and `whereis python`. Also whether invoking `python`, calls `python3.10`.

geoeo, 2024-02-07:
I checked all that. It was python3.10

cottsay, 2024-02-16:
When building an `ament_python` package, colcon will use whatever python interpreter it was built for. If you install it from debs, it should use the system's default interpreter. If you want it to use a different interpreter, you should install it using pip under that interpreter.

Building `ament_cmake` or `cmake` packages is a completely different story and really has nothing to do with colcon. The default logic of CMake's `find_package(Python3)` should discover the "latest" version of Python available which meets the criteria (has interpreter, libraries, etc).

To override which version of the Python interpreter you want CMake to use, pass `-DPython3_EXECUTABLE=/usr/bin/python3` to the CMake arguments for your build.