ros2/ros2 #1523: Closing the terminal doesn't kill the node in ROS2 Humble running on Docker

maxkonrad, 2024-02-13:
## Bug report

**Required Info:**

- Operating System:
  - Docker running Ubuntu 22.04
- Installation type:
  - via docker
- Version or commit hash:
  - ros:humble repo (using FROM ros:humble command in Dockerfile)
- DDS implementation:
  - idk about this I think it is default
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue
1- I connected to jetson nano host via ssh using my Ubuntu22.04 pc's Terminator terminal.
2- I ran a docker instance with the following Dockerfile
 `FROM ros:humble

RUN apt-get update && apt-get install -y nano && rm -rf /var/lib/apt/lists/*

COPY config/ /site_config/

ARG USERNAME=ros
ARG USER_UID=1000
ARG USER_GID=$USER_UID

RUN groupadd --gid $USER_GID $USERNAME \
  && useradd -s /bin/bash --uid $USER_UID --gid $USER_GID -m $USERNAME \
  && mkdir /home/$USERNAME/.config && chown $USER_UID:$USER_GID /home/$USERNAME/.config

RUN apt-get update \
  && apt-get install -y sudo \
  && echo $USERNAME ALL=\(root\) NOPASSWD:ALL > /etc/sudoers.d/$USERNAME\
  && chmod 0440 /etc/sudoers.d/$USERNAME \
  && rm -rf /var/lib/apt/lists/*

COPY entrypoint.sh /entrypoint.sh
COPY bashrc /home/$USERNAME/.bashrc

COPY /my_py_pkg /src/my_py_pkg

ENTRYPOINT [ "/bin/bash", "/entrypoint.sh" ]

CMD ["bash"]`

3- There are two std_msgs.msg int64 publishers I am using on the my_py_pkg python package, one of them publishes to /number_count topic and both of them publishes to /number topic. (idk if two nodes publishing to one topic is a problem)

4- Close the terminal or change the network.

#### Expected behavior
I expected running nodes to kill.

#### Actual behavior
Running nodes show when I run `ros2 node list` but when I run `ros2 lifecycle set <topic name> shutdown` it returns Node not found on terminal. I don't know if node is alive or not.

#### Additional information
[image]

fujitatomoya, 2024-02-13:
looks like network configuration for containers. it would be probably good if you can share the complete reproducible procedure to make it happen.

> I ran a docker instance with the following Dockerfile

can you share the exact command line how to start the container? `docker run <all options>`?

> change the network.

I am not following this. can you share what operation did you do to `change the network` and whose network?

> Running nodes show when I run ros2 node list but when I run ros2 lifecycle set <topic name> shutdown it returns Node not found on terminal.

probably DDS participant discovery completed so that you can see the ROS 2 context and nodes in the network, but endpoints such as topics and services.

audrow, 2024-02-22:
I copied this issue over to https://github.com/osrf/docker_images/issues/721, since it didn't seem that I can transfer issues between orgs. 

I'm going to close this issue. Thanks for posting @maxkonrad!

closed 2024-02-22 by audrow