ros2/ros2 #1531: ros2 daemon hangs on arch with docker

Sonic803, 2024-03-02:
## Bug report

**Required Info:**

- Operating System:
  - Arch Linux 6.1.38-1-lts
- Installation type:
  - Docker
- Version or commit hash:
  - humble

#### Steps to reproduce issue

Just run
```
docker run --rm -it ros:humble /bin/bash -c "ros2 topic list"
```

#### Expected behavior

Output something like 

```
/parameter_events
/rosout
```

#### Actual behavior

Gets stuck

#### Additional information

I tried the same thing on Ubuntu and there it works.

On arch I have
docker 1:25.0.3-1

tfoote, 2024-03-04:
I can confirm that it works on my debianish machine with docker `Docker version 20.10.21, build baeda1f`

I see that docker 25 is much newer, and potentially there's changes in docker's default behavior or how Arch's networking is setup. Arch isn't a standard development platform so we're going to need to rely on more info and debugging from you or other Arch users.

akssri-sony, 2024-03-06:
Can confirm it doesn't work humble- onwards (tested with iron/rolling); but foxy is fine.

I'm guessing this indicates some issue on the DDS implementation.

akssri-sony, 2024-03-06:
Temporary fix; call,
```
ulimit -n 1024
```
in the container before running ros2 cli stuff.

gavanderhoorn, 2024-03-07:
Reminds me of [Odd Docker roscore/roslaunch delays with Linux kernel 6.1](https://answers.ros.org/question/413528) on ROS Answers.

clalancette, 2024-03-07:
> Reminds me of [Odd Docker roscore/roslaunch delays with Linux kernel 6.1](https://answers.ros.org/question/413528) on ROS Answers.

Yep, very good point.  That also leads to https://stackoverflow.com/questions/899038/getting-the-highest-allocated-file-descriptor/918469#918469 and https://github.com/ruby/ruby/blob/v3_0_2/io.c#L6557-L6597 , which are interesting.

We could change this to try a (Linux-specific) hint, like that Ruby code (but obviously in Python).  Basically see if we can get the FDSize from `/proc`, and if so, use that as the hint on where to stop.  Otherwise, we would still fall back to the soft-limit.  This won't work for non-Linux, and won't work for systems where `/proc` isn't mounted, but those should be the minority of systems.

@akssri-sony If you'd like to give that a shot, I'd be happy to review it.

akssri-sony, 2024-03-08:
Thanks for the pointers @clalancette ! I've created a PR here,
https://github.com/ros2/ros2cli/pull/888

fujitatomoya, 2024-06-04:
@clalancette all backports are completed, can you close this? i do not have power to do that.

closed 2024-06-04 by clalancette