ros2/ros2 #1536: Subscription count is 0 after subscription is created

sim-lab-dev, 2024-03-27:
## Problem description
I have created two nodes for publishing and subscribing odometry message in ros foxy.There are about 6 different types of subscriber and 6 publisher in one node. The publishing node only publishes odometry message in 100 hz. But when the subscriber is created by rclcpp no message is received occasionally. By using "occasionally" means about 1/20 for subscriber to lost connection to publisher. And also when I use ros2 info to check the topic status, the subscription count is 0. So is there anybody knows why the subscriber can't receive message. 

**Required Info:**

- Operating System:
  - 
  - ubuntu20.04
- Installation type:
  - 
  - binaries
- Version or commit hash:
  - 
  - ros foxy
- DDS implementation:
  - 
  - default
- Client library (if applicable):
  -

sim-lab-dev, 2024-03-28:
> this is not much an information, but hopefully helpful.
> 
> * Can you use ROS Humble instead? Foxy is already E.O.L.
> * Can you restart the ros2 daemon with `ros2 daemon stop` and then `ros2 daemon start`, and `ros2 topic info xxx`? (if this finds the subscription, probably node graph in daemon is not updated.)

* Do you mean ROS Humble is more stable than Foxy?
* Restarting the ros2 daemon does not work

This [link](https://github.com/ros2/ros2/issues/1301) found Ros Foxy has a bug in topic message. Not sure if the problem is caused by the bug

fujitatomoya, 2024-03-28:
> Do you mean ROS Humble is more stable than Foxy?

yes, I do. Humble is maintained distro.

> This https://github.com/ros2/ros2/issues/1301 found Ros Foxy has a bug in topic message. Not sure if the problem is caused by the bug

foxy uses Fast-DDS `2.1.x`, https://github.com/eProsima/Fast-DDS/pull/2906 has been merged, i believe the final foxy release should include this bug fix.

sim-lab-dev, 2024-03-29:
> > Do you mean ROS Humble is more stable than Foxy?
> 
> yes, I do. Humble is maintained distro.
> 
> > This #1301 found Ros Foxy has a bug in topic message. Not sure if the problem is caused by the bug
> 
> foxy uses Fast-DDS `2.1.x`, [eProsima/Fast-DDS#2906](https://github.com/eProsima/Fast-DDS/pull/2906) has been merged, i believe the final foxy release should include this bug fix.

I found that the following two cases the messages are received normally:
* Config the fastdds to use UDP transport instead of Shared Memory
* Using the cyclone-dds instead of fast-dds
In the first case I guessed the buffer size was not large enough and I also tried to increase the segment size of the fast-dds in Shared Memory mode, but it didn't work.