ros2/ros2 #1538: Support topic instances in DDS topics

MiguelCompany, 2024-04-01:
## Feature request

Support processing of members annotated with `@key`

#### Feature description

Allowing the user to aggregate several topics into a single topic with several instances by leveraging the usage of `@key` annotation.

See [here](https://docs.vulcanexus.org/en/iron/rst/enhancements/keys/topic-keys.html) for motivation and proposed tutorials.

#### Implementation considerations

Related PRs:
- ros2/rosidl#
- ros2/rosidl_typesupport_fastrtps#
- ros2/rmw_fastrtps#

MiguelCompany, 2024-04-03:
@clalancette The PRs mentioned here are ready for review. Tests are passing locally with `--packages-up-to test_communication`

@asorbini @eboasson You might want to take a look and incorporate the necessary changes to your RMWs so we interoperate on keyed topics

MiguelCompany, 2024-04-08:
@ahcorde I rebased the first two PRs yesterday to get https://github.com/ros2/rosidl/pull/795. That should have fixed the linter failures from the runs on 2024-04-05.

ahcorde, 2024-04-08:
* Linux [image]
* Linux-aarch64 [image]
* Windows [image]
* Windows Debug [image]

bijoua29, 2024-12-21:
Just saw this issue and the work going on this area. Is this a core level feature that will be available for all RMWs? We are Connext users.

As long term DDS users, I'm really excited that this feature is finally making it into ros2 as it solves a lot of issues in our application where we use arbitrary depth QoS when having multiple instances of a topic as well an explosion of topics, one for each instance. 

Is the plan for this feature to be released for Kilted?

lobolanja, 2025-03-03:
Hi @MiguelCompany, @clalancette, @fujitatomoya,

What are the expectations on this? 

From RTI, we have already completed the work to support topic instances in our RMW (we will open the PR soon; it is already under internal review), and we would like to close this for the May release (Kilted Kaiyu).

fujitatomoya, 2025-03-03:
@lobolanja 

CC: @MiguelCompany @mjcarroll 

i would love to see this available for Kilted, almost related PRs are ready, i think we need to rebase some of them and start the CI.

bijoua29, 2025-03-04:
@fujitatomoya Any plans to add the @key annotation for msg files? Also, are there any changes in the ros2 cli to support keyed topics e.g. for 'ros2 topic echo' to echo a specific key?

fujitatomoya, 2025-03-04:
> Any plans to add the @key annotation for msg files?

i think this has been done by rosidl changes by @MiguelCompany 

> are there any changes in the ros2 cli to support keyed topics e.g. for 'ros2 topic echo' to echo a specific key?

could you describe this feature a bit? especially what's in it for user experience?

fujitatomoya, 2025-03-04:
@bijoua29 

> The documentation say the @key annotation is only supported for .idl files and not for .msg file.

you mean the documentation is https://github.com/ros2/design/pull/340, right?

@MiguelCompany @ahcorde is this true, i though https://github.com/ros2/rosidl/pull/796 enables key annotation in `.msg` file as well. or i may have mistaken something...

> As for the ros2 cli,, for a keyed topic I would want to echo the topic only for a specific key. That's the whole point of keyed topics

yeah, efficient filtering is one of the feature provided from keyed topic. but also, it improves efficiency in resource management and instance lifecycle tracking, and enables QoS configuration per instance to ensure data retention is managed separately for each key.

> Programmatically I presume you would do it with content filters but not sure if that's the case with the cli.

good question. currently we do not have `content filtering` feature via ros2cli. as you mentioned we can use this feature in the program to filter the some messages based on the message contents if rmw implementation supports the `content filtering`. we could expose the `content filtering` option for ros2cli subscription based command line interfaces, this could be useful. but i am concerned that some RMW tier 1 implementation does not support `content filtering` feature, that means those options are not reliable for the user. i am not sure if we would want to have an optional arguments that `sometimes works, sometimes does not` in ros2cli, it does not seem user-friendly experience for me... but thanks for bringing that up, it is interesting.

MiguelCompany, 2025-03-05:
> The documentation say the `@key` annotation is only supported for .idl files and not for .msg file.

_NB_: Please quote text with an @ so we don't accidentally mention a github user (he reacted with a confused emoji)

@bijoua29 @fujitatomoya Regarding this, annotations are currently supported on IDL only.
There is a command line tool called `rosidl` that allows for a `.msg / .srv / .action` file to be converted to its equivalent `.idl` file.

```
rosidl translate [-h] [-o PATH] [--use TRANSLATOR] --to FORMAT [--from FORMAT] [-I PATH]
                        package_name interface_file [interface_file ...]
```

For example, from my_package root folder:
```
rosidl translate --to idl my_package msg/Data.msg msg/OtherData.msg
```

MiguelCompany, 2025-03-06:
I do think there would be value in supporting annotations in `.msg` / `.srv` / `.action` files. Not only `key`, but also `optional`, `external`, `non-serialized`, or `id`, but there is currently no plan for it.

bijoua29, 2025-03-11:
Hmm. That's unfortunate. So, to use keys, we would need to idl files instead of msg files

fgallegosalido, 2025-03-18:
@MiguelCompany we've just opened a pull request with support for topic instances in the Connext RMW. Can you modify the description to add the pull request to the list? The pull request is [rmw_connextdds #178](https://github.com/ros2/rmw_connextdds/pull/178).

Do you need help rebasing the branches related to the main task? Mainly [rcl_interfaces #162](https://github.com/ros2/rcl_interfaces/pull/162) and [system_tests #542](https://github.com/ros2/system_tests/pull/542).

fujitatomoya, 2025-04-01:
@fgallegosalido @lobolanja @MiguelCompany 

we had a talk about this today's PMC meeting.
the question is that if we have IDL with key, does this break other RMW implementation like cyclone or zenoh? assuming that keys are ignored by rmw implementations do not support topic instance, we should explicitly add the note about `key` behavior is dependent on the rmw implementation?

MiguelCompany, 2025-04-02:
@fujitatomoya 
Publishers and subscriptions of DDS RMW implementations that differ in the support of the key annotation should not match (I already checked this with `rmw_cyclonedds_cpp` <-> `rmw_fastrtps_cpp`)

Other RMW implementations can decide what to do with the annotation. I don't expect them to fail.

MiguelCompany, 2025-04-03:
https://github.com/ros2/ros2/issues/1538#issuecomment-2772212868
https://github.com/ros2/ros2/issues/1538#issuecomment-2775684213

CI: https://github.com/ros2/rmw_fastrtps/pull/753#issuecomment-2774731861

@fujitatomoya @mjcarroll @clalancette 
FYI

MiguelCompany, 2025-04-07:
CI with [this repos file](https://gist.github.com/MiguelCompany/7cd1c280606e974132ad7ed9baf6c599/raw/6d6d058fcaa05a20ab811bbc94b48610ad31bbc3/ros2-topic-instances.repos): 
* Linux [image]
* Linux-aarch64 [image]
* Linux-rhel [image]
* Windows [image]

fujitatomoya, 2025-04-07:
@fgallegosalido @lobolanja @MiguelCompany the other thing i was thinking is that how do we want to document or explain this to the ROS 2 user or developers once those are merged? so the IDL files support `key` attribution, and a couple of rmw implementations (rmw_fastrtps and rmw_connextdds) can mange the `TopicInstance` by key, it would be useful to add more documentation like `what's in it for user application`?

- adding general introduction about topic instances and tutorial how to use it?
- user can reduce the number of endpoints, keyed topics allows us to publish data for multiple entities (like sensors, vehicles, users) under the same topic, distinguished by their keys. (can be beneficial for endpoint discovery cost.)
- per topic instance QoS history, liveliness life cycle management. (History QoS like KEEP_LAST or KEEP_ALL applies per instance, not per topic.)
- content filtering with topic instance, subscribers can apply content filters to only receive data for specific keys.

i am just not sure if we say this in the official documentation yet. because user needs to use `idl` file, but the ros2 documentation explains about `msg` files for type interfaces. on the other hand, `TopicInstance` can be really useful and solution against the situation described above... i maybe missing something, so i would like to hear opinion here.

CC: @wjwwood @ahcorde @mjcarroll @christophebedard @sloretz @cottsay

lobolanja, 2025-04-07:
> [@fgallegosalido](https://github.com/fgallegosalido) [@lobolanja](https://github.com/lobolanja) [@MiguelCompany](https://github.com/MiguelCompany) the other thing i was thinking is that how do we [...]

@fujitatomoya,

As far as I understand—since we joined the RTI side of this project just a few months ago—the idea is to address this in a second phase, where the RMW layer would need to introduce additional functionality in its APIs to fully support Topic Instances usage from the perspective of RMW users.

This PR is mainly focused on enabling the underlying support at the DDS level, but making full use of Topic Instances through the ROS 2 layers will require collaborative design and alignment between the ROS RMW committee and the vendors implementing RMW layers

MiguelCompany, 2025-04-07:
> the other thing i was thinking is that how do we want to document or explain this to the ROS 2 user or developers once those are merged? so the IDL files support `key` attribution, and a couple of [...]

@fujitatomoya We can prepare the documentation starting from [this](https://docs.vulcanexus.org/en/jazzy/rst/tutorials/core/deployment/keys/topic_keys.html) and [this](https://docs.vulcanexus.org/en/jazzy/rst/tutorials/core/deployment/keys/filtered_topic_keys.html) tutorials.

We should explain that on eProsima's and RTI's rmw layers, certain QoS are applied by instance. We can add a section explaining how to use this starting from a msg file, since there are tools available to convert a msg into an idl, so the user does not need to create the IDL from scratch.

bijoua29, 2025-04-27:
> As far as I understand—since we joined the RTI side of this project just a few months ago—the idea is to address this in a second phase, where the RMW layer would need to introduce additional functionality in its APIs to fully support Topic Instances usage from the perspective of RMW users.

> This PR is mainly focused on enabling the underlying support at the DDS level, but making full use of Topic Instances through the ROS 2 layers will require collaborative design and alignment between the ROS RMW committee and the vendors implementing RMW layers

I don't understand. I thought keyed topics are available for use with these changes, at least for fastrtps and connext, as long as you annotate the idl files. Is that not correct?

fujitatomoya, 2025-04-28:
@bijoua29 

> I don't understand. I thought keyed topics are available for use with these changes, at least for fastrtps and connext, as long as you annotate the idl files. Is that not correct?

yes, it is. i think what @lobolanja meant is that more generic APIs to take advantage of topic instances from ROS 2 application.

MiguelCompany, 2025-05-26:
Now that we have support for topic instances in rolling and kilted, including tutorials [here](https://docs.ros.org/en/kilted/Tutorials/Advanced/Topic-Keys/Topic-Keys-Tutorial.html) and [here](https://docs.ros.org/en/kilted/Tutorials/Advanced/Topic-Keys/Filtered-Topic-Keys-Tutorial.html), I think we can close this.

closed 2025-05-26 by MiguelCompany