ros2/ros2 #1539: Error with SetBool service

balfaroo, 2024-04-03:
**Required Info:**

- Operating System:
  - Ubuntu 22.04
- Installation type:
  - Source
- Version or commit hash:
  - Humble
- DDS implementation:
  - 
- Client library (if applicable):
  - rclpy

I am having an issue where I am trying to setup a SetBool service. In my code, I try to set the message field of the response as response.message = "success", but I am met with a TypeError() message. I am not sure why this is happening, any insights would be appreciated.

fujitatomoya, 2024-04-03:
> I am trying to setup a SetBool service. In my code, I try to set the message field of the response as response.message = "success", but I am met with a TypeError() message.

because you are trying to set string type to bool field?

i suggest that you can take a look at https://docs.ros.org/en/humble/Tutorials/Beginner-Client-Libraries/Writing-A-Simple-Py-Service-And-Client.html

balfaroo, 2024-04-03:
> because you are trying to set string type to bool field?
>
> i suggest that you can take a look at [...]

No, that's not the case since the response for SetBool contains a boolean field and a string field. The boolean field is being set fine, the string field is the one I'm having trouble with. I even set it to another data type to see what happened and I got another error that said message field expected type 'str'.

fujitatomoya, 2024-04-03:
can you share [Short, Self Contained, Correct (Compilable), Example](http://www.sscce.org/) to reproduce the issue?

balfaroo, 2024-04-03:
> can you share [Short, Self Contained, Correct (Compilable), Example](http://www.sscce.org/) to reproduce the issue?

Sure, here is the code in the client that is relevant to the service:

`
from std_srvs.srv import SetBool
# within constructor for node:
self.fly_cli = self.create_client(SetBool, 'arming_service', qos_profile=qos_profile)
self.req = SetBool.Request()

# within  a timer callback that checks if the service is needed:
self.req.data = True
self.future = self.fly_cli.call_async(self.req)
rclpy.spin_until_future_complete(self, self.future)
 print(self.future.message)

`

Here is the relevant code in the server:
`
from std_srvs.srv import SetBool

# within node initilalization:
self.arm_srv = self.create_service(SetBool, 'arming_service', self.arm, qos_profile=qos_profile)

# arm() callback function:
 def arm(self, request, response):
        """Send an arm command to the vehicle."""
        self.publish_vehicle_command(
            VehicleCommand.VEHICLE_CMD_COMPONENT_ARM_DISARM, param1=1.0)
        self.get_logger().info('Arm command sent')
        response.success = True
        print(type("arm command send"))
        response.message = "success"
    
        self.armed = True
`