ros2/ros2 #1541: typo in error message: "Be aware that are nodes ..."

VladimirFokow, 2024-04-17:
Required info:

- Operating System:
  -  Ubuntu 22.04
- Installation type:
  - binaries
- Version or commit hash:
  - humble-desktop
- DDS implementation:
  - cyclonedds-cpp

#### Steps to reproduce issue

- create 2 nodes that share the same name

- run `ros2 node list`

#### Expected behavior

no typo in the error message: "there are"

#### Actual behavior

typo: "that are"

#### Additional information

Example of the error message that I'm talking about can be found [across github](https://github.com/search?q=%22Be+aware+that+are+nodes+in+the+graph%22&type=code)

fujitatomoya, 2024-04-17:
this has been fixed with https://github.com/ros2/ros2cli/pull/775

closed 2024-04-17 by VladimirFokow

clalancette, 2024-04-17:
> @fujitatomoya oh, so this is not the kind of change to be backported to [humble](https://github.com/ros2/ros2cli/blob/humble/ros2node/ros2node/verb/list.py#L45) as I understand?

Yeah, this one could possibly break anything downstream that is expecting that exact string.  For that reason, we won't backport this to Humble.