ros2/ros2 #1542: ImportError from rqt_bag

leizhenyu-lzy, 2024-04-18:
```bash
$ rqt_bag 
RosPluginProvider.load(rqt_bag_plugins/BagImagePlugin) exception raised in __builtin__.__import__(rqt_bag_plugins.image_plugin, [ImagePlugin]):
Traceback (most recent call last):
  File "/opt/ros/humble/lib/python3.10/site-packages/rqt_gui/ros_plugin_provider.py", line 80, in load
    module = __builtin__.__import__(
  File "/opt/ros/humble/lib/python3.10/site-packages/rqt_bag_plugins/image_plugin.py", line 35, in <module>
    from .image_timeline_renderer import ImageTimelineRenderer
  File "/opt/ros/humble/lib/python3.10/site-packages/rqt_bag_plugins/image_timeline_renderer.py", line 44, in <module>
    from PIL.ImageQt import ImageQt
ImportError: cannot import name 'ImageQt' from 'PIL.ImageQt' (/home/lzy/.local/lib/python3.10/site-packages/PIL/ImageQt.py)

rqt_bag.TimelineFrame.load_plugins() failed to load plugin "rqt_bag_plugins/BagImagePlugin":
cannot import name 'ImageQt' from 'PIL.ImageQt' (/home/lzy/.local/lib/python3.10/site-packages/PIL/ImageQt.py)
```

```python
# maybe 
from PIL.ImageQt import ImageQt 
# can be changed to
from PIL import ImageQt
# ?
```

fujitatomoya, 2024-04-18:
that works for me with `rolling` https://github.com/ros2/ros2/commit/4d07f58591920794676261a14c2ba9d0a7bf2746

i think you are missing some required packages (Qt), and that leads https://github.com/python-pillow/Pillow/blob/f8160b858aebf8b7bffdefba48735feac300a060/src/PIL/ImageQt.py#L172 this if statement to be false?

there are no platform and distribution information on the issue header though.

MichaelOrlov, 2024-04-25:
@leizhenyu-lzy A friendly ping to follow up on the @fujitatomoya question 
> i think you are missing some required packages (Qt), and that leads https://github.com/python-pillow/Pillow/blob/f8160b858aebf8b7bffdefba48735feac300a060/src/PIL/ImageQt.py#L172 this if statement to be false?

andreasBihlmaier, 2024-05-02:
Same issue on jammy with latest humble packages with ros-humble-desktop-full, ros-humble-ros-gz-sim and ros-humble-ros-ign-gazebo (among other humble packages) installed.

devvaibhav455, 2024-09-27:
> Same issue on jammy with latest humble packages with ros-humble-desktop-full, ros-humble-ros-gz-sim and ros-humble-ros-ign-gazebo (among other humble packages) installed.
> 
> Solved by cleaning up my `~/.local/lib/python3.10`.

I uninstalled PIL and re-installed. It worked for me.

clalancette, 2026-10-26:
Looks like this is an environmental issue, so closing.  If you are still having issues, please feel free to reopen.

closed 2026-10-26 by clalancette