ros2/ros2 #1547: Can't set double_array parameter via ros2 param set  (humble)

AntonioViscomi, 2024-04-19:
ROS 2 Distro : Humble

## Bug report:
Can't set DOUBLE_ARRAY parameter via <code>ros2 param set node_name, parameter_name, value</code>

- Operating System:
Ubuntu 22.04
- Installation type:
Debian packages
- Version or commit hash:
  - 
- DDS implementation:
  - 
- Client library (if applicable):
  - 

#### Steps to reproduce issue

```python3
    #Define a parameter callback inside a node:
    def parameters_callback(self, params):

        for param in params:

            if param.name == 'control.feedback_gain':
                if param.type_ == Parameter.Type.DOUBLE_ARRAY:
                    self.set_parameters([Parameter('control.feedback_gain', 
                                                  Parameter.Type.DOUBLE_ARRAY, 
                                                  param.value)])
                
            else:
                return SetParametersResult(successful=False)
            
        return SetParametersResult(successful=True)
```
Then try to set it via bash command (mock values for debugging): 
```bash
ros2 param set vehicle_controller_node  control.feedback_gain [1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0,  1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0]
```
#### Expected behavior
Parameter set successful

#### Actual behavior

```bash
ros2: error: unrecognized arguments: 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0]
```
If I try to use the same syntax using integers I get:
```bash
Setting parameter failed: Wrong parameter type, expected 'Type.DOUBLE_ARRAY' got 'Type.INTEGER_ARRAY'
```
#### Additional information
I've seen another issue marked as solved, but I could not figure out a solution by myself

TheRealSalvo, 2024-04-22:
I replicated the problem on my machine. I solved removing the spaces in your command. Try:
```
ros2 param set vehicle_controller_node  control.feedback_gain [1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0,1.0]
```

TheRealSalvo, 2024-04-23:
Makes sense, without the double quotation the cli interpreted "[1.0,"  as input and the rest of the array as another input who could not recognise.