ros2/ros2 #1549: ROS2 Humble fails to build on Windows (mcap_vendor: missing: zstd_LIBRARY)

CraigBuilds, 2024-04-22:
## Bug report

**Required Info:**

- Operating System:
  - Windows 10
- Installation type:
  - Source
- DDS implementation:
  - RTI Connext

#### Steps to reproduce issue

```
colcon build --merge-install --packages-select mcap_vendor --cmake-clean-cache
```

#### Expected behavior

Builds

#### Actual behavior

```
Starting >>> mcap_vendor
--- stderr: mcap_vendor
CMake Error at C:/Program Files/CMake/share/cmake-3.21/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find zstd (missing: zstd_LIBRARY) (found version "1.5.6")
Call Stack (most recent call first):
  C:/Program Files/CMake/share/cmake-3.21/Modules/FindPackageHandleStandardArgs.cmake:594 (_FPHSA_FAILURE_MESSAGE)
  C:/dev/ros2_humble/install/share/zstd_vendor/cmake/Modules/Findzstd.cmake:36 (find_package_handle_standard_args)
  CMakeLists.txt:7 (find_package)

---
Failed   <<< mcap_vendor [10.1s, exited with code 1]

Summary: 0 packages finished [11.9s]
  1 package failed: mcap_vendor
  1 package had stderr output: mcap_vendor
```

christophebedard, 2024-04-22:
Can you try:

```
$ colcon build --merge-install --packages-up-to mcap_vendor --cmake-clean-cache
```

CraigBuilds, 2024-04-23:
Thank you for your reply. I tried this, and I received the same error:

```
...
Starting >>> ament_cmake
Finished <<< ament_cmake [7.25s]
Starting >>> zstd_vendor
[Processing: zstd_vendor]
Finished <<< zstd_vendor [45.3s]
Starting >>> mcap_vendor
--- stderr: mcap_vendor
CMake Error at C:/Program Files/CMake/share/cmake-3.21/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find zstd (missing: zstd_LIBRARY) (found version "1.5.6")
Call Stack (most recent call first):
  C:/Program Files/CMake/share/cmake-3.21/Modules/FindPackageHandleStandardArgs.cmake:594 (_FPHSA_FAILURE_MESSAGE)
  C:/dev/ros2_humble/install/share/zstd_vendor/cmake/Modules/Findzstd.cmake:36 (find_package_handle_standard_args)
  CMakeLists.txt:7 (find_package)

---
Failed   <<< mcap_vendor [9.92s, exited with code 1]

Summary: 20 packages finished [2min 18s]
  1 package failed: mcap_vendor
  1 package had stderr output: mcap_vendor
```

Note, during my initial attempts to fix this, I installed Zstandard 1.5.6 via chocolatey. Should I remove this?

CraigBuilds, 2024-04-23:
I tried from a fresh shell (which did not have the zstd environment variables pointing towards the chocolatey install, and now I get this output:

```
Starting >>> zstd_vendor
Finished <<< zstd_vendor [10.5s]
Starting >>> mcap_vendor
--- stderr: mcap_vendor
CMake Error: The current CMakeCache.txt directory C:/dev/ros2_humble/build/mcap_vendor/_deps/mcap-subbuild/CMakeCache.txt is different than the directory c:/dev/ros2_humble_src/build/mcap_vendor/_deps/mcap-subbuild where CMakeCache.txt was created. This may result in binaries being created in the wrong place. If you are not sure, reedit the CMakeCache.txt

CMake Error at C:/Program Files/CMake/share/cmake-3.21/Modules/FetchContent.cmake:1005 (message):
  CMake step for mcap failed: 1
Call Stack (most recent call first):
  C:/Program Files/CMake/share/cmake-3.21/Modules/FetchContent.cmake:1146:EVAL:2 (__FetchContent_directPopulate)
  C:/Program Files/CMake/share/cmake-3.21/Modules/FetchContent.cmake:1146 (cmake_language)
  C:/Program Files/CMake/share/cmake-3.21/Modules/FetchContent.cmake:1189 (FetchContent_Populate)
  CMakeLists.txt:38 (fetchcontent_makeavailable)
  CMakeLists.txt:81 (build_mcap_vendor)

---
Failed   <<< mcap_vendor [9.25s, exited with code 1]
```

The folder was named `ros2_humble_src` during one of the original builds, and then it was named to `ros2_humble`. I think this may be the new issue.

I used the `--cmake-clean-cache` flag. I'd rather not build everything from scratch as it takes a while. Is there anything else I can try?

CraigBuilds, 2024-04-23:
I started from scratch by cloning the `humble-release` branch and running `vcs import` on the local `.repos` file.
Prior to this, I was using `vcs import --input https://raw.githubusercontent.com/ros2/ros2/humble/ros2.repos src` as per the instructions.

It is now working. I'm unure why it was failing earlier.

closed 2024-04-23 by clalancette