ros2/ros2 #1551: Could not find a package configuration file provided by "ament_cmake"

Daneto96, 2024-04-24:
#### Required Info:

- Operating System:
  - Ubuntu 20.04.6 LTS

- Installation type:
  - ROS2-Humble following the offical documentation-> https://docs.ros.org/en/humble/Installation/Alternatives/Ubuntu-Development-Setup.html
- Purpose: Trying to install crazyswarm2 from bitcraze --> https://imrclab.github.io/crazyswarm2/installation.html

#### Additional information:

1. Looks quite similar to https://github.com/ros2/ros2/issues/1236 and https://github.com/ros2/ros2/issues/710
2. we tried to run the commands via sudo
3. we tried to run the commands as super user
4. we executed setup.bash & setup.sh (/home/bitcraze/ros2_humble/install/)

#### Expected behavior
succesfull execution of this command: sudo colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release

#### Actual behavior
[colconbuild_error.txt](https://github.com/ros2/ros2/files/15096484/colconbuild_error.txt)

christophebedard, 2024-04-24:
You should not need to use `sudo`.

Please ask a question on https://robotics.stackexchange.com/questions/tagged/ros. Feel free to link to it here. Make sure to include all the commands you ran, in the order that you ran them.

clalancette, 2024-05-02:
This looks very much like you have either too new or too old a version of pycryptography installed compared to your base system.  You should look into what version of pycryptography is installed, and what is needed by the version of python you have installed.

Because this doesn't look like a problem in ROS 2 (and should be asked over on stackexchange anyway), closing.

closed 2024-05-02 by clalancette