ros2/ros2 #1554: I recently encountered an issue while trying to install ROS2 Rolling to my Raspberry Pi 5 with Ubuntu 24.04 Noble Numbat LTS.

aarambhpatil, 2024-04-29:
## Bug report

**Required Info:**

- Operating System:
  - 
- Installation type:
  - 
- Version or commit hash:
  - 
- DDS implementation:
  - 
- Client library (if applicable):
  - 

#### Steps to reproduce issue

```

```

#### Expected behavior

#### Actual behavior

#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

clalancette, 2024-04-29:
> I recently encountered an issue while trying to install ROS2 Rolling to my Raspberry Pi 5 with Ubuntu 24.04 Noble Numbat LTS. There is a missing package in Ubuntu 24.04 named libpcl-common1.13 as it is not installed. The package ros-rolling-pcl-conversions depends on this specific package to be installed and it cannot be installed from the apt sources. Would be glad if anyone has found any solutions to the issue

It is fixed in `ros2-testing`; we'll be doing an update to fix it shortly.

clalancette, 2024-05-01:
This was fixed in the latest sync, so closing.

closed 2024-05-01 by clalancette