ros2/ros2 #1556: ROS2 python talker using tutorial_interfaces (Num type) causes segmentation fault

jgaitondef, 2024-04-29:
## Bug report

**Required Info:**

- Operating System:
  - 
  - Distributor ID:	Ubuntu
    Description:	Ubuntu 22.04.4 LTS
    Release:	22.04
    Codename:	jammy

- Installation type:
  - 
  - binaries
  - Ubuntu apt install
  
- Version or commit hash:
  - 
  - ROS2 Iron
  
- DDS implementation:
  - 
  - fastDDS
  
- Client library (if applicable):
  - 
  rclpy

#### Steps to reproduce issue

1. Created a python pub/sub using this link https://docs.ros.org/en/foxy/Tutorials/Beginner-Client-Libraries/Writing-A-Simple-Py-Publisher-And-Subscriber.html#
   This app worked for "String" type.
2. Create custom message types using tutorial interfaces  using link   https://docs.ros.org/en/foxy/Tutorials/Beginner-Client-Libraries/Custom-ROS2-Interfaces.html
3. Modified packages.xml of py_pubsub package to depend on tutorial interfaces so that we can use the newly defined custom interfaces Num.msg in particular

<?xml version="1.0"?>
<?xml-model href="http://download.ros.org/schema/package_format3.xsd" schematypens="http://www.w3.org/2001/XMLSchema"?>
<package format="3">
  <name>py_pubsub</name>
  <version>0.0.0</version>
  <description>DLP poc</description>
  <maintainer email="xyz@sample.com">xyz</maintainer>
  <license>Apache-2.0</license>

  <exec_depend>rclpy</exec_depend>
  <exec_depend>std_msgs</exec_depend>
  <exec_depend>tutorial_interfaces</exec_depend>

  <exec_depend>geometry_msgs</exec_depend>
  <exec_depend>rosidl_default_runtime</exec_depend>

  <test_depend>ament_copyright</test_depend>
  <test_depend>ament_flake8</test_depend>
  <test_depend>ament_pep257</test_depend>
  <test_depend>python3-pytest</test_depend>

  <export>
    <build_type>ament_python</build_type>
  </export>
</package>
~            

#### Expected behavior

- Run both talker and listener.  Expecting them to recognize the Num type and produce/ consume messages

#### Actual behavior

- Talker is exiting with Segmentation fault on executing line " self.publisher_.publish(msg)" in publisher class.

- ros2 run py_pubsub talker
[ros2run]: Segmentation fault

Any help is appreciated. Thanks

fujitatomoya, 2024-04-30:
> Created a python pub/sub using this link https://docs.ros.org/en/foxy/Tutorials/Beginner-Client-Libraries/Writing-A-Simple-Py-Publisher-And-Subscriber.html#
> Create custom message types using tutorial interfaces using link https://docs.ros.org/en/foxy/Tutorials/Beginner-Client-Libraries/Custom-ROS2-Interfaces.html

can you use the followings instead? since you are on iron distro.

- https://docs.ros.org/en/iron/Tutorials/Beginner-Client-Libraries/Writing-A-Simple-Py-Publisher-And-Subscriber.html
- https://docs.ros.org/en/iron/Tutorials/Beginner-Client-Libraries/Custom-ROS2-Interfaces.html

if that does not fix the segmentation fault problem, could you provide the [Short, Self Contained, Correct (Compilable), Example](http://www.sscce.org/) for iron branch?

jgaitondef, 2024-04-30:
Hi, 
  Thanks for the quick response and  inputs. I tried using the "iron" distros doc with no difference. 

  https://docs.ros.org/en/iron/Tutorials/Beginner-Client-Libraries/Custom-ROS2-Interfaces.html#testing-num-msg-with-pub-sub
   It is failing at the above point for both py_pubsub and py_srvcli examples (for the talker and srv client) with the usage of custom types Num.msg and AddThreeInts.srv  respectively. The segment fault is also traced to line where the newly defined custom type is published over the wire.

  The above link mentions modifying the CMakeLists.txt, however as I'm running python apps, there is no CMakeLists.txt file for both py_pubsub and py_srvcli. 

  Please note that the steps are strictly followed from the link and so is the code with no additional changes from my side.

  Thanks for your help.

jgaitondef, 2024-04-30:
Doctor report for your reference 

ros2 doctor --report

   NETWORK CONFIGURATION
inet         : 127.0.0.1
inet4        : ['127.0.0.1']
netmask      : 255.0.0.0
device       : lo
flags        : 73<UP,RUNNING,LOOPBACK>
mtu          : 65536
inet         : 172.24.141.33
inet4        : ['172.24.141.33']
ether        : c4:5a:b1:d7:f5:ae
netmask      : 255.255.252.0
device       : enp0s31f6
flags        : 4163<UP,RUNNING,MULTICAST,BROADCAST>
mtu          : 1500
broadcast    : 172.24.143.255
inet         : 192.168.122.1
inet4        : ['192.168.122.1']
ether        : 52:54:00:96:60:6d
netmask      : 255.255.255.0
device       : virbr0
flags        : 4099<UP,MULTICAST,BROADCAST>
mtu          : 1500
broadcast    : 192.168.122.255

   PACKAGE VERSIONS
tutorial_interfaces                       : latest=, local=0.0.0
py_srvcli                                 : latest=, local=0.0.0
py_pubsub                                 : latest=, local=0.0.0
launch_testing_examples                   : latest=0.18.0, local=0.18.0
examples_rclpy_pointcloud_publisher       : latest=0.18.0, local=0.18.0
examples_rclpy_minimal_subscriber         : latest=0.18.0, local=0.18.0
examples_rclpy_minimal_service            : latest=0.18.0, local=0.18.0
examples_rclpy_minimal_publisher          : latest=0.18.0, local=0.18.0
examples_rclpy_minimal_client             : latest=0.18.0, local=0.18.0
examples_rclpy_minimal_action_server      : latest=0.18.0, local=0.18.0
examples_rclpy_minimal_action_client      : latest=0.18.0, local=0.18.0
examples_rclpy_guard_conditions           : latest=0.18.0, local=0.18.0
examples_rclpy_executors                  : latest=0.18.0, local=0.18.0
examples_rclcpp_wait_set                  : latest=0.18.0, local=0.18.0
examples_rclcpp_multithreaded_executor    : latest=0.18.0, local=0.18.0
examples_rclcpp_minimal_timer             : latest=0.18.0, local=0.18.0
examples_rclcpp_minimal_subscriber        : latest=0.18.0, local=0.18.0
examples_rclcpp_minimal_service           : latest=0.18.0, local=0.18.0
examples_rclcpp_minimal_publisher         : latest=0.18.0, local=0.18.0
examples_rclcpp_minimal_composition       : latest=0.18.0, local=0.18.0
examples_rclcpp_minimal_client            : latest=0.18.0, local=0.18.0
examples_rclcpp_minimal_action_server     : latest=0.18.0, local=0.18.0
examples_rclcpp_minimal_action_client     : latest=0.18.0, local=0.18.0
examples_rclcpp_cbg_executor              : latest=0.18.0, local=0.18.0
examples_rclcpp_async_client              : latest=0.18.0, local=0.18.0
ros_environment                           : latest=4.1.1, local=4.1.1
tf2_py                                    : latest=0.31.6, local=0.31.6
ros2param                                 : latest=0.25.6, local=0.25.6
ament_cmake_export_include_directories    : latest=2.0.5, local=2.0.5
ament_lint_auto                           : latest=0.14.3, local=0.14.3
rosgraph_msgs                             : latest=1.6.0, local=1.6.0
sros2_cmake                               : latest=0.11.3, local=0.11.3
ament_flake8                              : latest=0.14.3, local=0.14.3
tf2_msgs                                  : latest=0.31.6, local=0.31.6
rclcpp_lifecycle                          : latest=21.0.6, local=21.0.6
rosidl_typesupport_fastrtps_cpp           : latest=3.0.2, local=3.0.2
ament_cmake_lint_cmake                    : latest=0.14.3, local=0.14.3
std_msgs                                  : latest=5.0.0, local=5.0.0
rosidl_adapter                            : latest=4.0.1, local=4.0.1
tf2_eigen_kdl                             : latest=0.31.6, local=0.31.6
launch_xml                                : latest=2.0.3, local=2.0.3
[... 262 lines cut ...]
rosbag2_transport                         : latest=0.22.6, local=0.22.6
rqt_bag_plugins                           : latest=1.3.4, local=1.3.4

   PLATFORM INFORMATION
system           : Linux
platform info    : Linux-6.5.0-28-generic-x86_64-with-glibc2.35
release          : 6.5.0-28-generic
processor        : x86_64

   QOS COMPATIBILITY LIST
compatibility status    : No publisher/subscriber pairs found

   RMW MIDDLEWARE
middleware name    : rmw_fastrtps_cpp

   ROS 2 INFORMATION
distribution name      : iron
distribution type      : ros2
distribution status    : active
release platforms      : {'rhel': ['9'], 'ubuntu': ['jammy']}

   TOPIC LIST
topic               : none
publisher count     : 0
subscriber count    : 0

jgaitondef, 2024-04-30:
ros2 doctor
/opt/ros/iron/lib/python3.10/site-packages/ros2doctor/api/package.py: 112: UserWarning: controller_manager_msgs has been updated to a new version. local: 3.24.0 < latest: 3.25.0
/opt/ros/iron/lib/python3.10/site-packages/ros2doctor/api/package.py: 112: UserWarning: rqt_joint_trajectory_controller has been updated to a new version. local: 3.22.0 < latest: 3.23.0
/opt/ros/iron/lib/python3.10/site-packages/ros2doctor/api/package.py: 112: UserWarning: rqt_controller_manager has been updated to a new version. local: 3.24.0 < latest: 3.25.0
/opt/ros/iron/lib/python3.10/site-packages/ros2doctor/api/package.py: 124: UserWarning: Cannot find the latest versions of packages:  tutorial_interfaces py_srvcli py_pubsub

All 5 checks passed

jgaitondef, 2024-05-01:
ros2 topic  pub topic tutorial_interfaces/msg/Num 'num: 1'
publisher: beginning loop
publishing #1: tutorial_interfaces.msg.Num(num=1)

Segmentation fault

Publishing the custom type with ros2 tool is also causing Seg fault

jgaitondef, 2024-05-03:
This issue was not observed on the "Rolling Ridley" distribution.