ros2/ros2 #1557: Ros2 sample Creating an action fails to compile

jgaitondef, 2024-04-30:
## Bug report

**Required Info:**

- Operating System:
  - 
  - Distributor ID: Ubuntu
    Description: Ubuntu 22.04.4 LTS
    Release: 22.04
    Codename: jammy
  
- Installation type:
  - 
  - binaries
  - apt install
  -
- Version or commit hash:
  - 
  - ROS2 iron
  
- DDS implementation:
  - 
  - fastDDS
  - 
- Client library (if applicable):
  - 
  - rclpy

#### Steps to reproduce issue

jgaitondef, 2024-04-30:
[0.231s] WARNING:colcon.colcon_core.package_selection:Some selected packages are already built in one or more underlay workspaces:
	'action_tutorials_interfaces' is in: /home/test/programs/ros2_ws/install/action_tutorials_interfaces, /opt/ros/iron
If a package in a merged underlay workspace is overridden and it installs headers, then all packages in the overlay must sort their include directories by workspace order. Failure to do so may result in build failures or undefined behavior at run time.
If the overridden package is used by another package in any underlay, then the overriding package in the overlay must be API and ABI compatible or undefined behavior at run time may occur.

If you understand the risks and want to override a package anyways, add the following to the command line:
	--allow-overriding action_tutorials_interfaces

This may be promoted to an error in a future release of colcon-override-check.
Starting >>> action_tutorials_interfaces
--- stderr: action_tutorials_interfaces                             
/home/test/programs/ros2_ws/src/action_tutorials_interfaces/build/action_tutorials_interfaces/rosidl_typesupport_fastrtps_c/action_tutorials_interfaces/action/detail/fibonacci__type_support_c.cpp: In function ‘bool _Fibonacci_Result__cdr_serialize(const void*, eprosima::fastcdr::Cdr&)’:
/home/test/programs/ros2_ws/src/action_tutorials_interfaces/build/action_tutorials_interfaces/rosidl_typesupport_fastrtps_c/action_tutorials_interfaces/action/detail/fibonacci__type_support_c.cpp:256:9: error: ‘class eprosima::fastcdr::Cdr’ has no member named ‘serializeArray’; did you mean ‘serialize_array’?
  256 |     cdr.serializeArray(array_ptr, size);
      |         ^~~~~~~~~~~~~~
      |         serialize_array
/home/test/programs/ros2_ws/src/action_tutorials_interfaces/build/action_tutorials_interfaces/rosidl_typesupport_fastrtps_c/action_tutorials_interfaces/action/detail/fibonacci__type_support_c.cpp: In function ‘bool _Fibonacci_Result__cdr_deserialize(eprosima::fastcdr::Cdr&, void*)’:
/home/test/programs/ros2_ws/src/action_tutorials_interfaces/build/action_tutorials_interfaces/rosidl_typesupport_fastrtps_c/action_tutorials_interfaces/action/detail/fibonacci__type_support_c.cpp:284:9: error: ‘class eprosima::fastcdr::Cdr’ has no member named ‘deserializeArray’; did you mean ‘deserialize_array’?
  284 |     cdr.deserializeArray(array_ptr, size);
      |         ^~~~~~~~~~~~~~~~
      |         deserialize_array
/home/test/programs/ros2_ws/src/action_tutorials_interfaces/build/action_tutorials_interfaces/rosidl_typesupport_fastrtps_c/action_tutorials_interfaces/action/detail/fibonacci__type_support_c.cpp: In function ‘bool _Fibonacci_Feedback__cdr_serialize(const void*, eprosima::fastcdr::Cdr&)’:
/home/test/programs/ros2_ws/src/action_tutorials_interfaces/build/action_tutorials_interfaces/rosidl_typesupport_fastrtps_c/action_tutorials_interfaces/action/detail/fibonacci__type_support_c.cpp:480:9: error: ‘class eprosima::fastcdr::Cdr’ has no member named ‘serializeArray’; did you mean ‘serialize_array’?
  480 |     cdr.serializeArray(array_ptr, size);
      |         ^~~~~~~~~~~~~~
      |         serialize_array
/home/test/programs/ros2_ws/src/action_tutorials_interfaces/build/action_tutorials_interfaces/rosidl_typesupport_fastrtps_c/action_tutorials_interfaces/action/detail/fibonacci__type_support_c.cpp: In function ‘bool _Fibonacci_Feedback__cdr_deserialize(eprosima::fastcdr::Cdr&, void*)’:
/home/test/programs/ros2_ws/src/action_tutorials_interfaces/build/action_tutorials_interfaces/rosidl_typesupport_fastrtps_c/action_tutorials_interfaces/action/detail/fibonacci__type_support_c.cpp:508:9: error: ‘class eprosima::fastcdr::Cdr’ has no member named ‘deserializeArray’; did you mean ‘deserialize_array’?
  508 |     cdr.deserializeArray(array_ptr, size);
      |         ^~~~~~~~~~~~~~~~
      |         deserialize_array
gmake[2]: *** [CMakeFiles/action_tutorials_interfaces__rosidl_typesupport_fastrtps_c.dir/build.make:99: CMakeFiles/action_tutorials_interfaces__rosidl_typesupport_fastrtps_c.dir/rosidl_typesupport_fastrtps_c/action_tutorials_interfaces/action/detail/fibonacci__type_support_c.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:267: CMakeFiles/action_tutorials_interfaces__rosidl_typesupport_fastrtps_c.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< action_tutorials_interfaces [3.28s, exited with code 2]

Summary: 0 packages finished [3.46s]
  1 package failed: action_tutorials_interfaces
  1 package had stderr output: action_tutorials_interfaces

christophebedard, 2024-04-30:
You're trying to build `action_tutorials_interfaces`, but it looks like it's already installed via `apt`/binaries. You're probably not using the right branch of the https://github.com/ros2/demos repository: make sure you're using the `iron` branch. Otherwise, since you `action_tutorials_interfaces` is already installed, you can remove it from your workspace to avoid building it.

If that doesn't work, please ask a question here: https://robotics.stackexchange.com/questions/tagged/ros. Include information all the commands you ran before `colcon build` and the code you're trying to build. Feel free to post the link to your question here.

closed 2024-04-30 by clalancette

jgaitondef, 2024-04-30:
ros2 doctor
/opt/ros/iron/lib/python3.10/site-packages/ros2doctor/api/package.py: 112: UserWarning: controller_manager_msgs has been updated to a new version. local: 3.24.0 < latest: 3.25.0
/opt/ros/iron/lib/python3.10/site-packages/ros2doctor/api/package.py: 112: UserWarning: rqt_joint_trajectory_controller has been updated to a new version. local: 3.22.0 < latest: 3.23.0
/opt/ros/iron/lib/python3.10/site-packages/ros2doctor/api/package.py: 112: UserWarning: rqt_controller_manager has been updated to a new version. local: 3.24.0 < latest: 3.25.0
/opt/ros/iron/lib/python3.10/site-packages/ros2doctor/api/package.py: 124: UserWarning: Cannot find the latest versions of packages:  tutorial_interfaces py_srvcli py_pubsub

All 5 checks passed

jgaitondef, 2024-04-30:
Doctor report for your reference

ros2 doctor --report

   NETWORK CONFIGURATION
inet         : 127.0.0.1
inet4        : ['127.0.0.1']
netmask      : 255.0.0.0
device       : lo
flags        : 73<UP,RUNNING,LOOPBACK>
mtu          : 65536
inet         : 172.24.141.33
inet4        : ['172.24.141.33']
ether        : c4:5a:b1:d7:f5:ae
netmask      : 255.255.252.0
device       : enp0s31f6
flags        : 4163<UP,RUNNING,MULTICAST,BROADCAST>
mtu          : 1500
broadcast    : 172.24.143.255
inet         : 192.168.122.1
inet4        : ['192.168.122.1']
ether        : 52:54:00:96:60:6d
netmask      : 255.255.255.0
device       : virbr0
flags        : 4099<UP,MULTICAST,BROADCAST>
mtu          : 1500
broadcast    : 192.168.122.255

   PACKAGE VERSIONS
tutorial_interfaces                       : latest=, local=0.0.0
py_srvcli                                 : latest=, local=0.0.0
py_pubsub                                 : latest=, local=0.0.0
launch_testing_examples                   : latest=0.18.0, local=0.18.0
examples_rclpy_pointcloud_publisher       : latest=0.18.0, local=0.18.0
examples_rclpy_minimal_subscriber         : latest=0.18.0, local=0.18.0
examples_rclpy_minimal_service            : latest=0.18.0, local=0.18.0
examples_rclpy_minimal_publisher          : latest=0.18.0, local=0.18.0
examples_rclpy_minimal_client             : latest=0.18.0, local=0.18.0
examples_rclpy_minimal_action_server      : latest=0.18.0, local=0.18.0
examples_rclpy_minimal_action_client      : latest=0.18.0, local=0.18.0
examples_rclpy_guard_conditions           : latest=0.18.0, local=0.18.0
examples_rclpy_executors                  : latest=0.18.0, local=0.18.0
examples_rclcpp_wait_set                  : latest=0.18.0, local=0.18.0
examples_rclcpp_multithreaded_executor    : latest=0.18.0, local=0.18.0
examples_rclcpp_minimal_timer             : latest=0.18.0, local=0.18.0
examples_rclcpp_minimal_subscriber        : latest=0.18.0, local=0.18.0
examples_rclcpp_minimal_service           : latest=0.18.0, local=0.18.0
examples_rclcpp_minimal_publisher         : latest=0.18.0, local=0.18.0
examples_rclcpp_minimal_composition       : latest=0.18.0, local=0.18.0
examples_rclcpp_minimal_client            : latest=0.18.0, local=0.18.0
examples_rclcpp_minimal_action_server     : latest=0.18.0, local=0.18.0
examples_rclcpp_minimal_action_client     : latest=0.18.0, local=0.18.0
examples_rclcpp_cbg_executor              : latest=0.18.0, local=0.18.0
examples_rclcpp_async_client              : latest=0.18.0, local=0.18.0
ros_environment                           : latest=4.1.1, local=4.1.1
tf2_py                                    : latest=0.31.6, local=0.31.6
ros2param                                 : latest=0.25.6, local=0.25.6
ament_cmake_export_include_directories    : latest=2.0.5, local=2.0.5
ament_lint_auto                           : latest=0.14.3, local=0.14.3
rosgraph_msgs                             : latest=1.6.0, local=1.6.0
sros2_cmake                               : latest=0.11.3, local=0.11.3
ament_flake8                              : latest=0.14.3, local=0.14.3
tf2_msgs                                  : latest=0.31.6, local=0.31.6
rclcpp_lifecycle                          : latest=21.0.6, local=21.0.6
rosidl_typesupport_fastrtps_cpp           : latest=3.0.2, local=3.0.2
ament_cmake_lint_cmake                    : latest=0.14.3, local=0.14.3
std_msgs                                  : latest=5.0.0, local=5.0.0
rosidl_adapter                            : latest=4.0.1, local=4.0.1
tf2_eigen_kdl                             : latest=0.31.6, local=0.31.6
launch_xml                                : latest=2.0.3, local=2.0.3
[... 262 lines cut ...]
rosbag2_transport                         : latest=0.22.6, local=0.22.6
rqt_bag_plugins                           : latest=1.3.4, local=1.3.4

   PLATFORM INFORMATION
system           : Linux
platform info    : Linux-6.5.0-28-generic-x86_64-with-glibc2.35
release          : 6.5.0-28-generic
processor        : x86_64

   QOS COMPATIBILITY LIST
compatibility status    : No publisher/subscriber pairs found

   RMW MIDDLEWARE
middleware name    : rmw_fastrtps_cpp

   ROS 2 INFORMATION
distribution name      : iron
distribution type      : ros2
distribution status    : active
release platforms      : {'rhel': ['9'], 'ubuntu': ['jammy']}

   TOPIC LIST
topic               : none
publisher count     : 0
subscriber count    : 0