ros2/ros2 #1558: Some components in jazzy-release point to branches and not tags

thomasmoore-torc, 2024-05-07:
## Bug report

**Required Info:**

- Operating System:
  - N/A
- Installation type:
  - Source
- Version or commit hash:
  - 1c4fa4f07ee18c966e0df8165c26b910657c2eb043bde443a1c7da1ddb5f268b
- DDS implementation:
  - N/A
- Client library (if applicable):
  - N/A

#### Details

The following repository versions in the `ros2.repos` file in the `jazzy-release` branch and `release-jazzy-beta-20240430` tag are for branches within those repositories and not tags:

* eProsima/Fast-CDR
  * Current version: 2.2.1
  * Correct version: v2.2.1
* eProsima/Fast-DDS
  * Current version: 2.14.0
  * Correct version: v2.14.0
* eclipse-iceoryx/iceoryx
  * Current version: 2.0.5
  * Correct version: v2.0.5

fujitatomoya, 2024-05-07:
@thomasmoore-torc thanks!

> eProsima/Fast-CDR
> Current version: 2.2.1
> Correct version: v2.2.1

tag `v2.2.1` is on branch `2.2.1`'s head.

> eProsima/Fast-DDS
> Current version: 2.14.0
> Correct version: v2.14.0

tag `v2.14.0` is on branch `2.14.0`'s head.

but,

> eclipse-iceoryx/iceoryx
>Current version: 2.0.5

no such branch nor tag.

>Correct version: v2.0.5

this should be used.

above understanding is correct, right?

fujitatomoya, 2024-05-07:
@thomasmoore-torc that would be really appreciated if you can review https://github.com/ros2/ros2/pull/1559!

fujitatomoya, 2024-05-07:
@clalancette @marcoag can you guys also take a look at this issue and PR?

thomasmoore-torc, 2024-05-07:
@fujitatomoya - the PR looks good to me. Thanks!

marcoag, 2024-05-07:
Good catch, thanks for the PR @fujitatomoya!

clalancette, 2024-05-10:
This was fixed in #1559, so closing.

closed 2024-05-10 by clalancette