ros2/ros2 #1562: QoS combability is too strict, should be more user-friendly and flexible?

fujitatomoya, 2024-05-10:
This is to open the discussion on [current ROS 2 QoS compatibility design behavior](https://docs.ros.org/en/rolling/Concepts/Intermediate/About-Quality-of-Service-Settings.html#qos-compatibilities), but bug report.

### Description

[current ROS 2 QoS compatibility design behavior](https://docs.ros.org/en/rolling/Concepts/Intermediate/About-Quality-of-Service-Settings.html#qos-compatibilities) requires strict match for `transient local`, either DataWriter(Publisher) offers or DataReader(Subscription) requests.
This is not user-application friendly, because once mismatch is detected, it will fail to create endpoints. that said, application needs to know what QoS is offered or requested before creating endpoints.

Basically this behavior comes from the DDS which prefers precise and strict QoS matching as RMW implementation.
But as ROS 2 and RMW, this needs to be reconsidered to provide user-friendly behavior.

### Consideration

just ideas, need to expand more.

- `fallback behavior`. even QoS is incompatible, for example publisher offers `volatile` but subscriber requests `transient local`, it should be able to communicate with `volatile` QoS offered by publisher. (maybe user should be warned that requested QoS cannot be set.)
- only receiver (subscriber) should request the QoS based on the requirement? in other words, publisher always set with most compatible QoS, and that is the receiver to decide. (I am not sure all QoS setting can be covered with this...)

### Additional information

This discussion is originally come from https://github.com/ros2/rclcpp/pull/2499

brennand, 2024-08-29:
100% Agree, I think ROS2 as default should allow any QoS to connect to each other. Maybe have a Flag that makes it a requirement to have the same QoS.  I have wasted so much time of the years tracking down these problems as there isn't any warning and the node will not subscribe to anything.  I have just spent the last couple of hours trying to figure out why a  image pipeline doesn't work, just to figure out that 1 of the plugins wasn't using Sensor QoS, but default.  no Error msgs, everything comes up fine, yet the image doesn't pass though all the nodes.

MiguelCompany, 2024-08-29:
> There were no error messages

There is at least [this warning](https://github.com/ros2/rclcpp/blob/ee94bc63e4ce47a502891480a2796b53d54fcdfc/rclcpp/src/rclcpp/subscription_base.cpp#L344), and the user can specify a callback to handle the Incompatible QoS events.

Also, the [Quality of Service settings](https://docs.ros.org/en/jazzy/Concepts/Intermediate/About-Quality-of-Service-Settings.html) documentation page informs about both the [QoS compatibilities](https://docs.ros.org/en/jazzy/Concepts/Intermediate/About-Quality-of-Service-Settings.html#qos-compatibilities) and the [QoS events](https://docs.ros.org/en/jazzy/Concepts/Intermediate/About-Quality-of-Service-Settings.html#qos-events)