ros2/ros2 #1563: Humble build failure on Mac(Sonoma14.4.1)

gijaeAhn, 2024-05-14:
## Bug report

**Required Info:**

- Operating System:
  - Sonoma 14.4.1
- Installation type:
  - from source
- Version or commit hash:
  - Build ROS2 as described here: https://docs.ros.org/en/humble/Installation/Alternatives/macOS-Development-Setup.html
- DDS implementation:
  - rti 7.3.0

Followed the instruction above in "pyenv" virtual environment ( 3.10.14)
```

#### Expected behavior
Build completed

#### Actual behavior

Build Failure

- Success packages before rviz_ogre_vendor

- Error output

--- stderr: rviz_ogre_vendor
patch: **** malformed patch at line 1817:
make[2]: *** [ogre-v1.12.1-prefix/src/ogre-v1.12.1-stamp/ogre-v1.12.1-patch] Error 2
make[1]: *** [CMakeFiles/ogre-v1.12.1.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< rviz_ogre_vendor [1.22s, exited with code 2]
Aborted  <<< ignition_cmake2_vendor [0.06s]
Aborted  <<< ament_cmake_pclint [0.07s]
Aborted  <<< ament_cmake_pycodestyle [0.06s]
Aborted  <<< ament_cmake_pyflakes [0.07s]
Aborted  <<< ament_cmake_auto [1.22s]
Aborted  <<< mcap_vendor [0.10s]
Aborted  <<< ament_uncrustify [1.43s]
Aborted  <<< launch_testing [2.56s]
Aborted  <<< cyclonedds [4.65s]
Aborted  <<< fastrtps [3.88s]

Summary: 83 packages finished [12.5s]
  1 package failed: rviz_ogre_vendor
  10 packages aborted: ament_cmake_auto ament_cmake_pclint ament_cmake_pycodestyle ament_cmake_pyflakes ament_uncrustify cyclonedds fastrtps ignition_cmake2_vendor launch_testing mcap_vendor
  31 packages had stderr output: ament_clang_format ament_clang_tidy ament_copyright ament_cppcheck ament_cpplint ament_flake8 ament_index_python ament_lint ament_lint_cmake ament_mypy ament_package ament_pclint ament_pep257 ament_pycodestyle ament_pyflakes ament_xmllint domain_coordinator fastcdr iceoryx_binding_c iceoryx_hoofs iceoryx_posh launch launch_testing launch_xml launch_yaml osrf_pycommon rosidl_cli rpyutils rviz_ogre_vendor tracetools_read tracetools_trace
  226 packages not processed

- Attempts
https://github.com/ros2/ros2/issues/700#issuecomment-496026773

I tried :
''' zsh
brew tap osrf/simulation
brew install ogre
'''

and got errors this :
==> cmake -DCMAKE_OSX_ARCHITECTURES='x86_64' -F/Library/Frameworks/ -lfreetype -DOGRE_BUILD_PLUGIN_CG=OFF ..
Last 15 lines from /Users/gj/Library/Logs/Homebrew/ogre/01.cmake:
-lfreetype
-DOGRE_BUILD_PLUGIN_CG=OFF
-DCMAKE_INSTALL_PREFIX=/opt/homebrew/Cellar/ogre/1.7.4
-DCMAKE_INSTALL_LIBDIR=lib
-DCMAKE_BUILD_TYPE=Release
-DCMAKE_FIND_FRAMEWORK=LAST
-DCMAKE_VERBOSE_MAKEFILE=ON
-DFETCHCONTENT_FULLY_DISCONNECTED=ON
-Wno-dev
-DBUILD_TESTING=OFF
-DCMAKE_OSX_SYSROOT=/Library/Developer/CommandLineTools/SDKs/MacOSX14.sdk
..

CMake Error: Unknown argument -lfreetype
CMake Error: Run 'cmake --help' for all supported options.

If reporting this issue please do so at (not Homebrew/brew or Homebrew/homebrew-core):
  https://github.com/osrf/homebrew-simulation/issues

These open issues may also help:
gz-harmonic failing to build on macOS arm, due to ogre failure https://github.com/osrf/homebrew-simulation/issues/2478

clalancette, 2024-05-14:
We only support macOS as Tier 3, which is community supported.  So we probably won't be looking into this.  However, if you figure out a way to fix it, we are happy to review patches that make it work.

Yadunund, 2024-05-14:
I'm assuming you're building the `humble` branch from source? This was fixed by https://github.com/ros2/rviz/issues/943 and is present on `rolling` and `iron`.

gijaeAhn, 2024-05-20:
> I'm assuming you're building the `humble` branch from source? This was fixed by [ros2/rviz#943](https://github.com/ros2/rviz/issues/943) and is present on `rolling` and `iron`.

I appreciate it. I'll give it a try.

Nyanziba, 2024-05-23:
In this repository, they patch this file so I can build ros2 https://github.com/TakanoTaiga/ros2_m1_native

woensug-choi, 2024-06-14:
https://github.com/TakanoTaiga/ros2_m1_native didn't work.. 
https://github.com/TakanoTaiga/ros2_m1_native/issues/13

gijaeAhn, 2024-07-04:
@woensug-choi 
Did you solve the problem??

woensug-choi, 2024-07-04:
@gijaeAhn made my own https://github.com/IOES-Lab/ROS2_Jazzy_MacOS_Native_AppleSilicon

deniscalin, 2024-09-14:
> @gijaeAhn made my own https://github.com/IOES-Lab/ROS2_Jazzy_MacOS_Native_AppleSilicon

Can confirm that this installation works for Sonoma 14.5 Apple Silicon M2 chip. I am running it now. This is amazing. I have spent close to a day trying to install through the official build-from-source on MacOS tutorials, working out which dependency versions should be used and with each run getting just a bit closer to the correct build, but not quite. It's not trivial as there are a few key dependency mismatches.

This repo does all of this for you by using the correct recent dependencies for the latest Jazzy version of Ros 2 + Gazebo Harmonic (haven't tested the Gazebo yet) native build on Apple Silicon.

Thank you for making this, @woensug-choi!!

woensug-choi, 2024-09-16:
@deniscalin Great! thank you for testing! If you need `mavros` try this too! https://github.com/IOES-Lab/ROS2_MAVROS_AppleSilicon

ericdusel-tri, 2024-10-21:
> @gijaeAhn made my own https://github.com/IOES-Lab/ROS2_Jazzy_MacOS_Native_AppleSilicon

Looks like it also works on macOS 15.0.1. I have a Mac book with M2 Max chip. I had to make the following change: 

              Also, had to specify `empy==3.3.4` in python packages installation to not get fails during building.

_Originally posted by @androbaza in https://github.com/IOES-Lab/ROS2_Jazzy_MacOS_Native_AppleSilicon/issues/2#issuecomment-2381292266_

Also I could NOT get it working on my Mac book air with M1.

woensug-choi, 2024-10-22:
@ericdusel-tri https://github.com/IOES-Lab/ROS2_Jazzy_MacOS_Native_AppleSilicon/pull/3 should fix empy version problem. Any trails why it failed on M1 Macbook air?

gijaeAhn, 2024-12-25:
@woensug-choi,

I followed your instructions for building ROS2 on macOS Sequia 15.2, but I encountered some build errors. Below are the details:

Summary of Issues:
	•	Failed Package: fastrtps
	•	Aborted Packages: gz_math_vendor, iceoryx_posh, rviz_ogre_vendor, uncrustify_vendor
	
`
1 error generated.
make[2]: *** [src/cpp/CMakeFiles/fastrtps.dir/rtps/writer/RTPSWriter.cpp.o] Error 1
make[1]: *** [src/cpp/CMakeFiles/fastrtps.dir/all] Error 2
make: *** [all] Error 2
make: INTERNAL: Exiting with 13 jobserver tokens available; should be 12!
`

`
Summary: 86 packages finished [1min 7s]
  1 package failed: fastrtps
  4 packages aborted: gz_math_vendor, iceoryx_posh, rviz_ogre_vendor, uncrustify_vendor
`

It appears that the error is nested within the fastrtps package and is related to the RTPSWriter.cpp file.
Could you help me if you know this one?

gijaeAhn, 2025-01-06:
@woensug-choi,

I resolved the issue by modifying the Boost dependency in Homebrew
- File: boost/interprocess/detail/utilities.hpp
- Change: Clang’s size type deduction (unsigned long) was causing errors, so I modified it to size_t.

This hack worked, and the build completed successfully. However, I’m worrying that simply changing this might cause other issues.

Dannylingen2, 2025-01-17:
@gijaeAhn  did this work for you long term also running into this issue

gijaeAhn, 2025-01-18:
> [@gijaeAhn](https://github.com/gijaeAhn) did this work for you long term also running into this issue
I’ve been using it without any issues so far.
If you are trying to build another library that depends on Boost, I recommend restoring it and proceeding with the build.

mjcarroll, 2025-03-29:
For MacOS support, these days, please also consider installing the dependencies via `pixi`, similar to how we do for Windows.  For more information, check out https://discourse.ros.org/t/upcoming-switch-of-windows-installation-to-pixi-conda/41916

This is also available for Humble and Jazzy.

closed 2025-03-29 by mjcarroll