ros2/ros2 #1564: feat: Converting between Ros Nodelist and Process Ids

wkaisertexas, 2024-05-15:
## Feature request

#### Feature description

Going from node name to process id(s).

I am building a profiling tool for ROS nodes and I can find out the nodes which are currently running and their topics. However, I am unable to find out which PID those node names correspond. 

Some nodes have the same executable and node name allowing `pgrep` to effectively filter out nodes by name. However, there does not seem to be an easy way to go from a ros node and namespace to a process ID. 

There was a way to do this in Ros 1 as there is a package which was [able to monitor CPU usage](https://github.com/alspitz/cpu_monitor). However, there seems to be no equivalent in Ros2.

Does anyone have *any* ideas on this?

clalancette, 2024-05-15:
> Some nodes have the same executable and node name allowing `pgrep` to effectively filter out nodes by name. However, there does not seem to be an easy way to go from a ros node and namespace to a process ID.

There are 2 main reasons we don't have this in ROS 2:

1.  There isn't a one-to-one correspondence between nodes and processes.  It is entirely possible (and very common) for multiple nodes to be in the same process.
2.  It isn't totally clear how to find out the process name in a cross-platform way, since we have to support Windows as well as Linux here.

wkaisertexas, 2024-05-15:
Process ids do not have to be unique, but it would be nice to know that mapping. 

Additionally, a good middle ground would be giving users easy access between the instance of a node and the executable which runs that node.

Currently, if two nodes with the same name exist, distinguishing between how they were launched is kind of pain.

Ros's commitment to cross platform support is admirable, but reducing to the lowest common feature set seems reductive.

fujitatomoya, 2024-05-16:
a couple of comments here,

- (besides @clalancette 's comment above) ROS 2 supports distributed application, that means there can be multiple systems running ROS 2 application. In this case, process id can be the same with different node name or namespace. having IP address with process ID is not good enough to tell the application either, when we use containers within the different pid namespace assigned. currently i am not sure how we want to (or should) support this...

@wkaisertexas

- i guess that what you want to do at this moment is like a top command dedicated to ROS 2 in localhost system? in that case, how about https://github.com/iwatake2222/rotop? i have never tried that, but it seems that it works on the localhost system.

christophebedard, 2024-05-16:
You could try to use [`ros2_tracing`](https://github.com/ros2/ros2_tracing) and process the [`rcl_node_init`](https://github.com/ros2/ros2_tracing/blob/c0e59c442e61f6c638f06a48a00ecf10913442fb/tracetools/include/tracetools/tracetools.h#L180-L195) trace events (which can include PID) [live](https://lttng.org/docs/v2.13/#doc-lttng-live).

wkaisertexas, 2024-05-16:
That sounds like a great solution. Thank you so much!

audrow, 2024-05-23:
It does seem like this would be useful, but it's not clear what this means when nodes are running on different machines. Some middlewares have host information, but I don't think all of them do, so there probably isn't a clean way to do this for different middlewares.

I'm going to close this issue. Feel free to continue the discussion or open it back up, if it makes sense to do so.

closed 2024-05-23 by audrow

wjwwood, 2024-08-09:
If you're using Fast-DDS they have a "physical data topic" that might be of use. Of course this won't show up as a ROS topic, but maybe you could use Fast-DDS directly or their tools: https://fast-dds.docs.eprosima.com/en/latest/fastdds/property_policies/non_consolidated_qos.html#physical-data-in-discovery-information

DDS has hashes of some of this information in their participant id, which we do have a map from nodes -> participant ids internally in the DDS based rmw implementations (but that info is not public), but it's not very useful for debugging, see:

https://community.rti.com/rti-doc/500/ndds.5.0.0/doc/html/api_cpp/structDDS__WireProtocolQosPolicy.html#a1ec4d6bea721b98a18f77a6f427071a0

Alternatively, we could provide an interface like this ourselves, but as @clalancette said, it would require working out some details and also considering things like security (i.e. is it safe to publicly share hostname? pid? maybe, but what about process name? or even potentially worse the process arguments, etc).

There is also the "network flow endpoint" API in rmw, which was added to help support configuring QoS for 5G connections, and it might include things like protocol (TCP vs UDP) and IP addr for a given publisher, but not the process id:

https://github.com/ros2/rmw/blob/ac4f9afccca3dc305f2b519fdb0fb9131250f2a5/rmw/include/rmw/network_flow_endpoint.h#L48-L67