ros2/ros2 #1567: ros-rolling-desktop-full is not fully compatible with Ubuntu 24.04 LTS

ionutnechita-intel, 2024-05-24:
No template. Is not necessary.

OS: Ubuntu 24.04 LTS with Kernel 6.9 Vanilla.

I put ROS2 repository in apt sources. And i try to install ros-rolling-desktop-full package. And i notice one behaviour regarding this package:

```
apt-get install ros-rolling-desktop-full
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 ros-rolling-desktop-full : Depends: ros-rolling-ros-ign-gazebo-demos but it is not installable
                            Depends: ros-rolling-simulation but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
```

BR,
Ionut Nechita

clalancette, 2024-05-24:
It looks like you have outdated metadata, as that should have been fixed a long time ago.  Can you run `sudo apt-cache policy ros-rolling-desktop-full` and paste the output here?

ionutnechita-intel, 2024-05-24:
I have this:
```
 //# sudo apt-cache policy ros-rolling-desktop-full
ros-rolling-desktop-full:
  Installed: (none)
  Candidate: 0.10.0-3noble.20240429.164020
  Version table:
     0.10.0-3noble.20240429.164020 500
        500 http://packages.ros.org/ros2/ubuntu noble/main amd64 Packages
```

clalancette, 2024-05-24:
The sync was [done today](https://discourse.ros.org/t/new-packages-for-ros-2-rolling-ridley-2024-05-24/37902), so closing.

ionutnechita-intel, 2024-05-27:
Thank you.
Now is completed on Ubuntu 24.04 LTS.

Enjoy with rolling and jazzy on Ubuntu 24.04 LTS.