ros2/ros2 #1569: Buffer overflow during rclpy.create_node in python

TadaKazuto, 2024-05-28:
Failed to create rclpy node with error message "buffer overflow detected ***: terminated" while maybe hundreds of processes(or threads) held index(they were not running any more though).

Probably some python thread control library is not designed to release index(or dequeue) unless explicitly call shutdown process of each thread and it produces buffer overflow when we create ros2 rclpy node.

I just would like to know if it's known issue or any improvement is planned in future releases.

fujitatomoya, 2024-05-28:
@TadaKazuto can you provide [Short, Self Contained, Correct (Compilable), Example](http://www.sscce.org/) for the issue?

TadaKazuto, 2024-06-02:
I could fix the issue with changing process executor context option to "spawn" (instead of "fork" which is set on default in linux)

```
from multiprocessing import get_context
ProcessPoolExecutor(mp_context=get_context("spawn"))
```
My guess is that number of created sub-process is piled up when we keep launching new process and it would cause ros2 rclpy buffer overflow(probably while threading or something) since the number of created sub-process is inherited with "fork" option. On the other hand, brand new process without the info is activated with "spawn", and can works as a solution for this.