ros2/ros2 #1572: Make issue templates consistent across repositories.

audrow, 2024-06-06:
A great example of what we could do for all repositories is https://github.com/ros2/launch/pull/782#pullrequestreview-2102677808.

One known issue with our current issue template is that they point to ROS Answers, rather than the Robotics Stack Exchange.

erickisos, 2024-09-20:
Right now, there is only one template. Do we want to have multiple ones, as they did in the shared example?

erickisos, 2024-09-20:
Nvm, the old template has two cases, one for bug report and one for feature request.

fujitatomoya, 2024-11-09:
@audrow can you take a look at,

- https://github.com/ros2/rclcpp/pull/2667
- https://github.com/ros2/rclpy/pull/1379

if they are good to go, i can open more PRs for other repositories with some cosmetic changes.

fujitatomoya, 2024-11-10:
There has been discussion on if we should have organization level issue templates instead of each repository.

- much better for maintenance cost.
- we can override the organization issue template with repository's one if necessary.

IMO, organization issue templates would be the good option. (see my comments, https://github.com/ros2/rclcpp/pull/2667)

gavanderhoorn, 2024-11-10:
> There has been discussion on if we should have organization level issue templates instead of each repository.

https://github.com/ros2/.github/issues/2?

fujitatomoya, 2024-11-10:
@gavanderhoorn i had no idea we have that issue already, i will open the for there, thanks 👍

closed 2025-03-25 by mjcarroll