ros2/ros2 #1573: Incorrect Windows binary names on Jazzy release

RangaSundarRS, 2024-06-10:
## Bug report

**Required Info:**

- Operating System: Windows 10
- Installation type: Binaries
- Version or commit hash: Jazzy 2024-05-23
- DDS implementation: N/A
- Client library (if applicable): N/A

#### Steps to reproduce issue
Open github.com/ros2/ros2/releases/tag/release-jazzy-20240523 and view the binary packages.

#### Expected behavior
Files should be named "ros2-jazzy-20240523-windows-release-amd64.zip" and "ros2-jazzy-20240523-windows-debug-amd64.zip"
#### Actual behavior
Files are named "ros2-jazzy-20250523-windows-release-amd64.zip" and "ros2-jazzy-20230523-windows-debug.amd64.zip"

[image]

Yadunund, 2024-06-10:
@marcoag do you mind re-uploading the windows zip file after fixing the year in the name?

marcoag, 2024-06-11:
Thanks for the catch @RangaSundarRS, names should be fixed now.

clalancette, 2024-06-11:
Yep, they look fixed now.  Thanks for reporting @RangaSundarRS

closed 2024-06-11 by clalancette