ros2/ros2 #1577: colcon build python package on Jazzy

yjphhw, 2024-07-07:
seems easy_install will be deprecated， is there any  solution in future?

The output message:

--- stderr: test_package_python                   
/usr/lib/python3/dist-packages/setuptools/command/develop.py:40: EasyInstallDeprecationWarning: easy_install command is deprecated.
!!

        ********************************************************************************
        Please avoid running ``setup.py`` and ``easy_install``.
        Instead, use pypa/build, pypa/installer or other
        standards-based tools.

        See https://github.com/pypa/setuptools/issues/917 for details.
        ********************************************************************************

!!
  easy_install.initialize_options(self)
---
Finished <<< test_package_python [1.30s]

horverno, 2024-10-16:
same problem here with a really basic python package

``` r 

--- stderr: time_utils
/usr/lib/python3/dist-packages/setuptools/command/develop.py:40: EasyInstallDeprecationWarning: easy_install command is deprecated.
!!

        ********************************************************************************
        Please avoid running ``setup.py`` and ``easy_install``.
        Instead, use pypa/build, pypa/installer or other
        standards-based tools.

        See https://github.com/pypa/setuptools/issues/917 for details.
        ********************************************************************************

!!
  easy_install.initialize_options(self)
---

```

christophebedard, 2024-10-16:
Those are just warnings and will probably not be changed/addressed for Ubuntu 22.04.

black-puppydog, 2024-10-25:
Came here to report the same. I'm overhauling a tutorial for a ROS 2 course for beginners, and beginners tend to think they did something wrong if they follow a tutorial and there's big warnings popping up. 🙂

Is there a tracking issue for a migration to a newer build/packaging system for rclpy etc?

fujitatomoya, 2024-10-25:
@black-puppydog 

related issues and discussion,

- https://github.com/colcon/colcon-core/issues/454
- https://discourse.ros.org/t/call-for-testing-standards-based-python-packaging-with-colcon/32008

clalancette, 2024-10-26:
Can someone give a full example of a package that causes these warnings?  Because by default, we don't use `easy_install` for `ament_python` packages, which makes me think something else is going on here.

clalancette, 2024-11-07:
Thanks for the reproducer, that is helpful.  It turns out that I don't see this locally, and we don't see this in CI, because this comes about because of the `--symlink-install` argument.  If I remove that argument, then the warning goes away.  I'm not totally sure what the difference is, but there are a couple of temporary workarounds you can use:

1.  Build without `--symlink-install`.  That means that you'll need to rebuild anytime you change anything.
2.  Suppress the warning yourself, by following something like https://robotics.stackexchange.com/questions/112857/easyinstalldeprecationwarning-easy-install-command-is-deprecated-with-ros2-jaz .  For me, I could get this to all work with:

```
$ PYTHONWARNINGS="ignore:easy_install command is deprecated,ignore:setup.py install is deprecated" colcon build --packages-select p_controller --symlink-install
```

black-puppydog, 2024-11-07:
@clalancette thanks, that incantation is actually gonna come in very handy since the big fat warning message tends to confuse especially newcomers.

However, you write in https://github.com/colcon/colcon-core/pull/676 that it would "fix" this issue. I'd be weary about that. From what you write, and the fact that the workaround does work, it seems like future versions of python will break the `--symlink-install` flow?

clalancette, 2024-11-08:
> However, you write in [colcon/colcon-core#676](https://github.com/colcon/colcon-core/pull/676) that it would "fix" this issue. I'd be weary about that. From what you write, and the fact that the workaround does work, it seems like future versions of python will break the `--symlink-install` flow?

Future versions of Python will likely break everything about Python packaging.  PyPA is aggressively making changes to setuptools/python packaging, with little regards for backwards compatibility.

@cottsay has done quite of bit of detailed investigation and work on this, which resulted in https://github.com/colcon/colcon-python-project .  However, we can't switch to that because in this brave new world, we can't make `--symlink-install` work.  That will require additional development, and/or changes to upstream Python packaging.

All of that is to say is that hiding the warning for `--symlink-install` is relatively "safe"; it should continue to work as long as current setuptools packaging works.

black-puppydog, 2024-11-11:
Thanks for the explanation, good to know it's being investigated in detail already. 🙂

sungpyo9053, 2026-10-30:
Update for anyone landing here in 2026: I re-measured this on the official `ros:jazzy` image (colcon-core 0.21.3 from apt, setuptools 68.1.2 from apt) with a minimal `ament_python` package.

| colcon-core | build | SetuptoolsDeprecationWarning | EasyInstallDeprecationWarning | `ros2 run` |
|---|---|---|---|---|
| 0.21.3 (apt, current image) | default | 0 | 0 | works |
| 0.21.3 (apt, current image) | `--symlink-install` | 0 | 0 | works |
| 0.15.2 (pip, old) | default | 1 | 0 | works |
| 0.15.2 (pip, old) | `--symlink-install` | 1 | 1 | works |

So the `--symlink-install` dependence @clalancette described reproduces with an old colcon-core, and the suppression from colcon/colcon-core#676 (released in 0.18.4) removes it. If you still see the warning, `sudo apt update && sudo apt install python3-colcon-core` is the fix; the build result was the same either way. Humble and Lyrical gave the same result for the current images.

Raw output and the repro script: https://github.com/sungpyo9053/blog/tree/main/editorial/en/2026-11-16-ros2-colcon-setuptools-deprecation-warning/evidence

Disclosure: I run huntlab.app, where these measurements will also appear in a post. This comment was drafted with an AI assistant and checked against the raw logs above.