ros2/ros2 #1578: ROS2 Humble rqt_graph Problem

leizhenyu-lzy, 2024-07-07:
## Bug report

**Required Info:**

- Operating System: 
  - Ubuntu 22.04.4 LTS x86_64
- Installation type: 
  - binaries
- Version or commit hash:
  - 
- DDS implementation: 
  -  FASTDDS
- Client library (if applicable):
  -  just demo_nodes

#### Steps to reproduce issue

```bash
ros2 run demo_nodes_py listener
ros2 run demo_nodes_cpp talker
rqt_graph 
ros2 topic list 
ros2 service list 
```

#### Expected behavior

no empty ellipse

#### Actual behavior

[image]

#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

leizhenyu-lzy, 2024-07-07:
I have re-installed the whole rqt through apt, but the bug remains.

leizhenyu-lzy, 2024-07-07:
Does that ellipse represent ros2cli daemon?

[image]
[image]

leizhenyu-lzy, 2024-07-07:
[image]

Vedhamshbode, 2025-01-15:
I tried with Humble. This is what I get. No abnormality.
[image]
[image]

christophebedard, 2025-01-23:
I tried with Humble (using binaries on Ubuntu 22.04) and I'm getting the same result as Tomoya.

[image]

Unchecking _debug_ shows the hidden rqt node:

[image]

I ran the following commands after taking the screenshots above, so that's why the hidden demon node isn't shown in `rqt_graph`:

```
$ ros2 topic list
/chatter
/parameter_events
/rosout

$ ros2 service list
/listener/describe_parameters
/listener/get_parameter_types
/listener/get_parameters
/listener/list_parameters
/listener/set_parameters
/listener/set_parameters_atomically
/rqt_gui_py_node_277794/describe_parameters
/rqt_gui_py_node_277794/get_parameter_types
/rqt_gui_py_node_277794/get_parameters
/rqt_gui_py_node_277794/list_parameters
/rqt_gui_py_node_277794/set_parameters
/rqt_gui_py_node_277794/set_parameters_atomically
/talker/describe_parameters
/talker/get_parameter_types
/talker/get_parameters
/talker/list_parameters
/talker/set_parameters
/talker/set_parameters_atomically

```

If you're still facing this issue, try launching `rqt_graph` with `--clear-config` and `--force-discover`:

```
$ rqt_graph --clear-config --force-discover
```

Yadunund, 2025-02-07:
@leizhenyu-lzy we discussed this issue and while it may be reproducible for some, we're hoping someone from the community can dig deeper into what causes this and submit a fix that we'd be happy to review.