ros2/ros2 #1579: ROS2 Foxy cotainer will crash in ROS2 Jazzy

rayee-github, 2024-07-09:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 24.04
- Installation type:
  - ROS2 Jazzy
- DDS implementation:
  - Fast-RTPS

#### Steps to reproduce issue

Terminial 1
```
docker run -it --rm osrf/ros:foxy-desktop ros2 run demo_nodes_cpp talker
```
Terminial 2
```
docker run -it --rm osrf/ros:foxy-desktop ros2 run demo_nodes_cpp listener
```
Terminial 3
```
ros2 topic list
ros2 topic list --no-daemon
ros2 topic echo /chatter
```

#### Expected behavior
Show topic correctly

#### Actual behavior
1. Container crashes
[image]
2. RAM gets quickly filled up, causing a crash.
https://github.com/ros2/ros2/assets/32994498/e129388d-5c3b-4389-b86a-39441deef8b9

#### Additional information
But if I use a Jazzy container, it works fine.
Why container in foxy and jazzy incompatible?
```
docker run -it --rm osrf/ros:jazzy-desktop ros2 run demo_nodes_cpp talker
docker run -it --rm osrf/ros:jazzy-desktop ros2 run demo_nodes_cpp listener
ros2 topic echo /chatter
```
[image]

gavanderhoorn, 2024-07-09:
I'm just another user, but: likely a binary incompatibility in the protocol versions used by Jazzy and Foxy.

And obligatory, but: there is no guarantee of (nor any work towards) cross-distribution compatibility. Some may find this unfortunate, but given the limited resources available, there's only so much people can do and additionally, having to keep things working across releases (almost) 5 years apart is just too restrictive (ie: would hamper development of new features).

For a more thorough discussion: [ROS Cross-Distribution Communication](https://discourse.ros.org/t/ros-cross-distribution-communication/27335) and [ROS2 Inter-distro Communication Issue: Eloquent to Humble with Fast DDS](https://discourse.ros.org/t/ros2-inter-distro-communication-issue-eloquent-to-humble-with-fast-dds/33550).

clalancette, 2024-07-09:
Yes, exactly as @gavanderhoorn says; there is no guarantee of cross-distribution compatibility, and indeed we changed things a lot between Foxy and Jazzy.

> (nor any work towards) cross-distribution compatibility.

This is not strictly true.  In Iron, we added in the first steps towards cross-distribution compatibility.  However, there is a lot of work left to get to true cross-distribution compatibility, and it isn't currently scheduled.

closed 2024-07-09 by clalancette

gavanderhoorn, 2024-07-09:
I'm only aware of the type hashes to prevent incompatible matches.

Are you including type adaptation features in the cross-distro compatibility discussion, or something else?

rayee-github, 2024-07-10:
@clalancette 
Thank you for your reply.

But what I care more about is that if the incompatibility causes the program or node to crash, it will not have much impact, but why the incompatibility will cause the RAM to be filled up.
This will have a major impact on the security of many existing applications, or some uninformed people.

fujitatomoya, 2024-07-18:
> Test on second computer: Even worse, RAM gets quickly filled up, causing a crash.
> https://github.com/ros2/ros2/assets/32994498/e129388d-5c3b-4389-b86a-39441deef8b9

this could be related to ros2 daemon process but i am not sure yet. this still uses cross-distribution, so unexpected behavior would happen.
it would be probably nice that cross-distribution should not communicate (even discovery) each other unless user specifies that to avoid this kind of unsupported situation especially using containers.