ros2/ros2 #1580: E: Unable to locate package ros-jazzy-desktop

MickeyTso, 2024-07-09:
The following error message is reported during installing ROS2 Jazzy on the development board: `E: Unable to locate package ros-jazzy-desktop`, what should I do?The installed system is Ubuntu 20.04.

gavanderhoorn, 2024-07-09:
> The installed system is Ubuntu 20.04.

please post these kinds of questions on Robotics Stack Exchange in the future, but just to help you out: Jazzy doesn't support Ubuntu `20.04`. Only Ubuntu Noble (ie: `24.04`).

MickeyTso, 2024-07-09:
My development board contacted the manufacturer and only supports 22.04, can I install the fox version of 20.04 and 22.04? Later, we will change to the latest version during the testing phase.

closed 2024-07-09 by clalancette