ros2/ros2 #1581: Inconsistent file extensions on Jazzy Patch Release 1

RangaSundarRS, 2024-07-09:
## Bug report

**Required Info:**

- Operating System:
  - N/A
- Installation type:
  - Binaries
- Version or commit hash:
  - Jazzy Patch Release 1 (20240705): commit a964556303a322b91de47e954824c18e627a4bda
- DDS implementation:
  - N/A
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue
The file names on the GitHub release for Jazzy Patch Release 1 are not consistent with previous releases. In this case, they are named *.zip.zip or *.tar.bz2.bz2. At least the Windows Release version appears to be only one zip file (not nested), so only a file name change should be required.

#### Expected behavior
Files have one extension, for example ros2-jazzy-20240705-linux-noble-amd64.tar.bz2 or ros2-jazzy-20240705-windows-release-amd64.zip. 
#### Actual behavior
Files' extensions are listed twice: ros2-jazzy-20240705-windows-release-amd64.zip.zip
[image]

#### Additional information
In the future, an automated system for file naming may be useful, as there have been file naming issues on both of Jazzy's releases so far (see #1573 ). This naming inconsistency makes it more challenging to have automated tools which interact with the releases.

RangaSundarRS, 2024-07-09:
@marcoag If you could help again that would be great!

marcoag, 2024-07-09:
Thanks for the catch @RangaSundarRS, names should be fixed now.

closed 2024-07-09 by marcoag