ros2/ros2 #1589: Security vulnerabilities due to incomplete privilege inheritance in ROS2 and ROS1.

LichenXia198, 2024-08-07:
Hi,
We would like to discuss two potential vulnerabilities due to incomplete privilege inheritance in ROS2 and ROS1. We believe these vulnerabilities can become a security threat in ROS projects involving ROS2 and ROS1 packages from different sources. We have conducted experiments to verify the vulnerabilities, but please do correct us if we are wrong.

## Details of the first vulnerability
- Operating System:
  - Ubuntu 22.04
- Version
  - ROS2 Humble
- DDS implementation:
  -  eProsima’s Fast DDS 
- Client library:
  - rclcpp and rclpy
## Description
We find that ROS2 nodes can create new contexts through exposed interfaces of ROS2 Client Libraries or the ROS2 command line interface (cli). The newly created context is completely blank, meaning it does not inherit settings (including Domain ID, Namespace, and SROS2) from the node creating it. As a result, there is a risk that malicious attackers could utilize this mechanism to bypass the enforced isolation and security methods (SROS2).
## Reproduction steps
Attack scenario:
The attacker has control over several nodes existing in a running ROS2 system that comprises ROS2 packages from different parties. The accessibility of the compromised nodes is regulated by Domain ID, Namespace, and SROS2. The attacker has knowledge of ROS2’s Client Libraries and ROS2cli commands and can exploit them through compromised nodes.

Steps:
Through either ROS2 Client Libraries or ROS2cli, the attacker can create and configure a new context. They can then create a new node and assign the context to the new node so it is not restricted by its parent node’s privileges. 

1. ROS2 Client Libraries: By using the context constructor _rclcpp::Context_ or _rclpy.context.Context_, the attacker can create a blank context and further configure it with the desired Domain ID, Namespace and SROS2 settings. Then the new context can be assigned to the newly created node through node constructor _rclcpp::Node_ or _rclpy.node.Node._
2. ROS2cli: Through _std::system()_, the attacker can launch a node with a new context by executing commands such as _“ros2 run <package> <node>”_. The context configuration can be passed to the ROS2 commands as arguments.

Manipulation of SROS2:
While configuring the newly created context, the settings of SROS2 can be manipulated. Firstly, the attacker can disable SROS2 enforcement, allowing new nodes to freely access unprotected topics. Secondly, the attacker can dynamically assign any leaked enclave to the new nodes.

## Potential mitigation
We recommend restricting ROS2 nodes from accessing the interfaces of ROS2 Client Libraries mentioned in the reproduction steps and the entire ROS2cli.

## Details of the second vulnerability
- Operating System:
  - Ubuntu 20.04
- Version
  - ROS2 Rolling and ROS1 Noetic
- DDS implementation:
  -  eProsima’s Fast DDS 
- Client library:
  - rclcpp
## Description
We found that ROS2 nodes can create new ROS1 nodes through ROS1 user-space APIs or the ROS1 command line interface (CLI). The newly created ROS1 nodes can freely access ROS1 topics, including those connected to ROS2 topics by the ROS1 bridge. Even if the ROS2 node is prevented from accessing the connected ROS2 topics, it can indirectly access those ROS2 topics through the corresponding ROS1 topics. As a result, there is a risk that malicious attackers could utilize the ROS1 bridge to bypass the enforced isolation (Domain ID and Namespace) and security methods (SROS2).
## Reproduction steps
Attack scenario:
The attacker has control over several nodes existing in a running ROS2 system that comprises ROS2 packages from different parties. At the same time, there is a ROS1 bridge that connects the ROS2 and ROS1 systems, and some ROS1 and ROS2 topics have been connected. The accessibility of the compromised nodes is regulated by Domain ID, Namespace, and SROS2. The attacker has knowledge of ROS1 user-space APIs and ROS1 CLI commands and can exploit them through compromised ROS2 nodes.

Steps:
Through either ROS1 user-space APIs or ROS1 CLI, the attacker can create a ROS1 node and then connect the node to the ROS1 topics that are connected to the ROS2 topics.

1. ROS1 user-space APIs: By using the ROS1 API _ros::init()_ and _ros::NodeHandle_, the new ROS1 node can be constructed by the attacker through compromised ROS2 nodes. Then the attacker can connect the ROS1 node to the targeted ROS1 topic 
2. ROS1cli: Through _std::system()_, the attacker can launch a ROS1 node by executing commands like _“rosrun <package> <node>”_ , or kill a ROS1 node through commands such as _“rosnode kill <node_name>”_.
## Potential mitigation
We recommend restricting ROS2 nodes from accessing the ROS1 user-space APIs and ROS1cli.

clalancette, 2024-08-07:
> Firstly, the attacker can disable SROS2 enforcement, allowing new nodes to freely access unprotected topics.

Correct, if you don't use SROS2, you can access all topics.  That is a feature of ROS 2.

> Secondly, the attacker can dynamically assign any leaked enclave to the new nodes.

I'm not sure exactly what this part means.

Can you please provide some example code that demonstrates the bug you are running into?  Without that, it will be difficult to determine whether this is a problem or not.

LichenXia198, 2024-08-07:
Thank you for your reply. I realize I may not have made my point clear. Currently, I am researching security issues while ROS2 is used as a platform for a multi-robot scenario (MRS) where robots from different parties are working together under a ROS environment. In this scenario, different robots may have different privileges and belong to different users or tenants, and I assume that the attackers have already controlled one or more nodes and are trying to launch attacks from their own robot to disrupt other robots.
Therefore, I wasn't referring to bugs or issues with SROS2 itself. Instead, I wanted to report vulnerabilities that ROS2 allows malicious nodes (controlled by attackers) to escalate their network isolation and SROS2 settings. Initially, I tried to report these vulnerabilities by directly sending an email to ros@osrfoundation.org. However, I was not allowed to post messages to this address, so I decided to open this issue.

Here is an example of the topic callback function from a malicious node used in our experiment:
```python
def topic_callback(self, msg):
    """
    The node tries to create new nodes in its topic callback
    """
    os.environ["ROS_SECURITY_ENABLE"] = false # Disable SROS2
    new_context = Context()
    rclpy.init(args=[], context=new_context, domain_id=5) # Create and initialize a new context with a new Domain ID and SROS2 disabled
    attackNode = DoorRequestNode('door_request_node', True, new_context) # Create a new node under the new context, so the new node is not restricted by SROS2.
    rclpy.spin(attackNode)
```

clalancette, 2024-08-07:
> Initially, I tried to report these vulnerabilities by directly sending an email to [ros@osrfoundation.org](mailto:ros@osrfoundation.org).

Ah, thanks for that notification.  You should be able to email to that address; can you please try again, and if it fails, post a copy of the error you receive?  That said, we can continue here.

> Therefore, I wasn't referring to bugs or issues with SROS2 itself. Instead, I wanted to report vulnerabilities that ROS2 allows malicious nodes (controlled by attackers) to escalate their network isolation and SROS2 settings.

But I still don't understand what you are getting at here.  Can you please define "escalate their network isolation and SROS2 settings"?

Yes, attackers have full control over their own processes.  And in a default ROS 2 network, any node can join the network.  That is by design in this system.  To mitigate that, people can either run SROS2, or they can run their ROS 2 network over a VPN.

What would really help here is a complete example, including code for some nodes, a launch file, and some explanation of exactly what your expectation is and what is being violated.

LichenXia198, 2024-08-07:
I tried to send an email again, but still failed, here is the error message I received:

Hello XXX,

We're writing to let you know that the group you tried to contact (ros) may not exist, or you may not have permission to post messages to the group. A few more details on why you weren't able to post:

 * You might have spelled or formatted the group name incorrectly.
 * The owner of the group may have removed this group.
 * You may need to join the group before receiving permission to post.
 * This group may not be open to posting.

If you have questions related to this or any other Google Group, visit the Help Center at https://support.google.com/a/osrfoundation.org/bin/topic.py?topic=25838.

Thanks,

[osrfoundation.org](http://osrfoundation.org/) admins

fujitatomoya, 2024-08-08:
>     rclpy.init(args=[], context=new_context, domain_id=5) # Create and initialize a new context with a new Domain ID and SROS2 disabled
>     attackNode = DoorRequestNode('door_request_node', True, new_context) # Create a new node under the new context, so the new node is not restricted by SROS2.

i believe that the network group with domain ID 5 should be protected by authentication and access permission by SROS2. in that case, `attackNode` cannot even discover or be discovered by nodes with domain ID 5. i am also not sure what the security problem here?

before mentioning above, probably network isolation would be the 1st good security barrier, and in that network we can use SROS2 for authentication and access permission so that we can have secured ROS 2 network in the application layer. based on the security strategy and requirement, we could have security observability or runtime enforcement if necessary. but i think those are not only for ROS 2 application but any application framework.

LichenXia198, 2024-08-08:
> What would really help here is a complete example, including code for some nodes, a launch file, and some explanation of exactly what your expectation is and what is being violated.

Here is an example of a node trying to create a child node in the 'topic_callback()' function. 
What is my expectation:
Even with the newly created context, the child node should be forced to have the same settings (e.g., Domain ID, SROS2 enclave) as its parent node. Or the context constructor should not be allowed to be called by user-space code, given that the context constructor is a method for internal use, as mentioned in the [document].(https://docs.ros.org/en/humble/p/rclcpp/generated/index.html). Additionally, the example code worked with eProsima Fast DDS but failed with Eclipse Cyclone DDS, indicating inconsistency between different DDS implementations in terms of the context constructor.

```python
class DoorRequestNode(Node):
    def __init__(self, name ,received, context, namespace=None):
        super().__init__(name, context=context, namespace=namespace, use_global_arguments=True)
        if not received:
            self.first_message_received = received
            self.subscription = self.create_subscription(
                String, 
                'loophole', 
                self.topic_callback, 
                10
            )

    
    def topic_callback(self, msg):
        """
        The node tries to create new nodes in its topic callback
        """
        os.environ["ROS_SECURITY_ENCLAVE_OVERRIDE"] = "/office/door_supervisor" # assign a new enclave to the new context
        new_context = Context()
        rclpy.init(args=None, context=new_context, domain_id=5) # Create and initialize a new context with a new Domain ID and a new SROS2 enclave
        attackNode = DoorRequestNode('door_request_node', True, new_context) # Create a new node under the new context, so the new node is under a new domain with the permission granted by the new enclave.
        rclpy.spin(attackNode)

def main(args=None):
    rclpy.init(args=['--ros-args', '--enclave', '/office/attacker']) # An enclave is assigned to the attacker-robot1 node while launching
    door_request_node = DoorRequestNode('attacker_robot1', False, None)
    rclpy.spin(door_request_node)

    door_request_node.destroy_node()
    rclpy.shutdown()

if __name__ == '__main__':
    main()

clalancette, 2024-08-08:
> Even with the newly created context, the child node should be forced to have the same settings (e.g., Domain ID, SROS2 enclave) as its parent node.

Can you explain why this is your expectation?  The nodes are not directly related; the fact that a second one is created from the callback of the first one really doesn't make them related, especially because they are in different Contexts.

Further, this entire process is under the attackers control.  Obviously the attacker can do arbitrary things from within their own process.  The real question is whether an attacker can cause problems on *other* parts of the ROS network.  The attacker causing issues within their own code may be a bug, but doesn't seem like a security issue.

In other words, I still don't understand your expectations.  A more complete example (with other nodes on the network), and more definition around why you have your expectations, would help.

gavanderhoorn, 2024-08-09:
From a more "OS-like" perspective, perhaps the understanding and expectation is that nodes are like processes: child processes inherit quite a few things from the process that spawned them, including permissions, access levels and (sometimes) security contexts (whatever that term means in such contexts).

Being able to manipulate a security context from within a corrupted/malicious process could then result in privilege escalation, which would obviously be (un)desirable (depending on who's side your own).

@LichenXia198: could you clarify whether this is along the lines of what you were thinking when analysing `rclcpp` and `rclpy`?

LichenXia198, 2024-08-09:
Yes, that’s exactly what I was thinking. The potential for privilege escalation through manipulation of security contexts within a compromised node is indeed the core concern I was addressing. Thanks for putting it so clearly!

fujitatomoya, 2024-08-09:
@gavanderhoorn @LichenXia198 

> Being able to manipulate a security context from within a corrupted/malicious process could then result in privilege escalation, which would obviously be (un)desirable (depending on who's side your own).

How does it `manipulate` the security context? cz i think it is not permitted from this process space to access the other security enclaves. with example provided here, it seems it just does not use the security context, i am not sure how unsecured app manipulates the security context... those are the same process space and permission, how can it access the security context that is used other apps and permission? i may be mistaken something??? 🤔

LichenXia198, 2024-08-09:
>those are the same process space and permission, how can it access the security context that is used other apps and permission? i may be mistaken something??? 🤔

Indeed, a node in one process cannot directly access the security context in another process.

> How does it `manipulate` the security context? cz i think it is not permitted from this process space to access the other security enclaves. 

This is at least not guaranteed by ROS2. For example, if a process has access to a file system, it can access all the security enclaves in that file system. Therefore, if two processes share a file system (which is common in operating systems) to store their security enclaves, there is a risk that one process may use the enclave belonging to the other process.

> with example provided here, it seems it just does not use the security context, i am not sure how unsecured app manipulates the security context... 

In my second example, I demonstrated that the node can assign a different enclave (a new security context) to its child node.

fujitatomoya, 2024-08-09:
> For example, if a process has access to a file system, it can access all the security enclaves in that file system.
> Therefore, if two processes share a file system (which is common in operating [...]

This is bad system design for security, it should not be accessed by different security groups. system administrator should set appropriate permission for the file system based on user or group identification. as far as i can see, this is not ROS 2 security issue at all. I will store the different security data in different access permission for the file system.

MichaelOrlov, 2024-08-12:
@LichenXia198 Using ros1bridge with SROS is not meant to be secure by design IMO.
What we can do in this case?
The ros1bridge does what it is meant to do. i.e. a getaway from ROS1 to ROS2.

LichenXia198, 2024-08-12:
In the specific scenario we were focusing on (a ROS2 node accessing a protected ROS2 topic through the ROS1-bridge by creating a ROS1 node), the initial solution is straightforward: ROS2 user-space code should generally not be able to use ROS1 APIs. For more general cases where ros1bridge is utilized by arbitrary attackers to access protected ROS2 topics, the solution becomes more complicated. One potential solution could be to enable authentication on the ROS1 side as well; as far as I know, there is an experimental version of SROS for ROS1. As you mentioned, ros1bridge performs as intended. Therefore, the first step is to make users aware of this risk (e.g., by mentioning it in the repository) so they can be more cautious when deploying ros1bridge in their projects.

LichenXia198, 2024-08-12:
> This is bad system design for security, it should not be accessed by different security groups. system administrator should set appropriate permission for the file system based on user or group [...]

This is just an example to illustrate that the ROS2 node can switch between different permissions by itself when multiple security enclaves are available. We were not concerned about how the enclave might be leaked; instead, we emphasized that the ROS2 node has the ability to utilize the leaked enclave through the mechanisms and methods provided by ROS2 (by creating child nodes).

LichenXia198, 2024-08-15:
@mjcarroll Thank you for the update. I believe you have understood most of our report, but I would like to clarify a few points.
> if the child node can access alternative secure enclaves, that would be a permissions issue on the enclaves themselves, outside of the ROS framework

For the first report, we do assume that a secure enclave has already been compromised in the attack scenario. However, we believe that ROS2 should not provide methods for a node to exploit the compromised enclave (by creating child nodes) to gain additional permissions. This issue arises because the child node, with its new context, does not inherit the permissions of its parent node.
As shown in the previous example, the node 'attacker_robot1' is initially assigned the enclave 'attacker' and only has access to the topic 'loophole.' However, by creating a child node 'door_request_node' and assigning it the leaked enclave 'door_supervisor,' the node 'attacker_robot1' can access the topic 'door_requests' through its child node.
> ```python
> class DoorRequestNode(Node):
>     def __init__(self, name ,received, context, namespace=None):
>         super().__init__(name, context=context, namespace=namespace, use_global_arguments=True)
>         if not received:
>             self.first_message_received = received
>             self.subscription = self.create_subscription(
>                 String, 
>                 'loophole', 
>                 self.topic_callback, 
>                 10
>             )
>         else:
>             self.publisher_ = self.create_publisher(DoorRequest, 'door_requests', 10)
>     def topic_callback(self, msg):
>         """
>         The node tries to create new nodes in its topic callback
>         """
>         os.environ["ROS_SECURITY_ENCLAVE_OVERRIDE"] = "/office/door_supervisor" # assign a new enclave to the new context
>         new_context = Context()
>         rclpy.init(args=None, context=new_context, domain_id=5) # Create and initialize a new context with a new Domain ID and a new SROS2 enclave
>         attackNode = DoorRequestNode('door_request_node', True, new_context) # Create a new node under the new context, so the new node is under a new domain with the permission granted by the new enclave.
>         rclpy.spin(attackNode)
> 
> def main(args=None):
>     rclpy.init(args=['--ros-args', '--enclave', '/office/attacker']) # An enclave is assigned to the attacker-robot1 node while launching
>     door_request_node = DoorRequestNode('attacker_robot1', False, None)
>     rclpy.spin(door_request_node)
> 
>     door_request_node.destroy_node()
>     rclpy.shutdown()
> 
> if __name__ == '__main__':
>     main()
> ```

> We were not concerned about how the enclave might be leaked; instead, we emphasized that the ROS2 node has the ability to utilize the leaked enclave through the mechanisms and methods provided by ROS2 (by creating child nodes).

fujitatomoya, 2024-08-18:
> This is just an example to illustrate that the ROS2 node can switch between different permissions by itself when multiple security enclaves are available.

So can you please the concrete example that you are trying to point out as security breach?
Please understand that I am not making excuses, just trying to understand your point.

And with your example above, I say that cannot be done if we set the file access permission appropriately.
With your example, multiple security enclaves are accessible because system administrator sets so.
Can you tell me how to switch the another security enclaves that are not accessible via file system?

> os.environ["ROS_SECURITY_ENCLAVE_OVERRIDE"] = "/office/door_supervisor" # assign a new enclave to the new context
>         new_context = Context()
>         rclpy.init(args=None, context=new_context, [...]

With the same example here, this will fail with permission error since that this process id and group cannot (should not) access this file system.
Probably an another concrete example would be better to help everyone understand the security problem for ROS 2.

> we believe that ROS2 should not provide methods for a node to exploit the compromised enclave (by creating child nodes)

BTW, i think you are creating another context, not child nodes under the same context. (be advised that there can be multiple contexts in the process space.)

> This issue arises because the child node, with its new context, does not inherit the permissions of its parent node.

So if the system administrator does not set the file system permission for security enclaves and all enclaves are exposed to anyone (system security is already compromised...), any contexts in process space can already use any security enclaves. 
Besides, very 1st `rclpy.init(context)` can be already compromised with another security enclaves by attackers, because this process can access the security data. why 2nd (or `child`) `rclpy.init(context)` is the point here?

IMO, this comes down to the access permission and system security for security enclaves.

clalancette, 2024-08-19:
> So can you please the concrete example that you are trying to point out as security breach?

At this point, this is the most important point.  Please setup a git repository somewhere that shows a complete example of the exploit, along with an explanation of exactly what is happening and what you think should be happening.  Ideally we should be able to clone that repository, and run one or two commands, and see the problem.

If you want to keep that example repository private, then please feel free to send a link to it to security@openrobotics.org, which is a private list and only has a few subscribers to it.

LichenXia198, 2024-08-20:
I packaged the code and enclaves from an experiment with a complete explanation at https://github.com/LichenXia198/ros2_security_example

gavanderhoorn, 2024-08-20:
From [LichenXia198/ros2_security_example/README.md](https://github.com/LichenXia198/ros2_security_example/blob/8e47270acd549ca9b9b045841d39ede2e574e0ba/README.md):
> In a secure OS, a process should not be allowed to create a child process that has more permissions than its parent by any means.

while I don't dismiss a potential problem with how things work has been identified, I believe this shows a misunderstanding or at least a difference in expectation.

> In a secure OS, [..]

If we want to be pedantic, ROS 2 is not a secure OS. According to some, it's not an OS, period.

IIRC, SROS2 only claims to secure node-to-node communication (using DDS Security). Nothing else. It's certainly not a *secure OS*.

> Similarly, if we view ROS2 from an 'OS-like' perspective and apply the same principle, a node should not be able to create a child node that has more permissions to topics (even if a new context is created for the node). This example clearly violates that principle.

The repository you share seems to create two stand-alone nodes each with access to different enclaves (on disk). The fact one is created 'from' the other does not change that. [this](https://github.com/LichenXia198/ros2_security_example/blob/8e47270acd549ca9b9b045841d39ede2e574e0ba/src/cpp_pubsub/src/listener_test.cpp#L25-L29) does not create a child in the 'child process' sense. It instantiates a C++ object from a method in another C++ object. All in the same process. Any parent-child relationship seems only implied.

I put *from* in quotes earlier as that's a bit of a stretch. There's no `fork(..)` or anything like that. No resources get duplicated (like file descriptors), no security descriptors, no access permissions are passed along or inherited (like a true *child process* in a POSIX OS would get).

AFAICT, there's also no implication (in the API docs or implementation: the word *child* does not appear in `rclcpp/node.hpp` fi) a `Node` created 'inside' or 'from' or 'by' another `Node` will have any special relationship to that other node. They're just two `Node` instances.

> While 'first_listener' is running, suppose another secure enclave, "talker2", is leaked to it. For example, an attacker might distribute the enclave from another device to the device on which the node 'first_listener' is running. **But how the enclave 'talker2' is leaked and distributed is actually not our concern**.

well .. in case you're using files on a file system to share secrets (certificates in this case), keeping those files, on that file system, secret is definitely a concern. It's a requirement for this implementation to remain secure.

I'm not a security researcher, but these sentences seem to say something similar to "how this private key got leaked is not our concern, but anyone with that private key can now use it -> that's a security breach". While it would amount to a breach, we can't claim openssl, ssh, gpg or anything using that private key is compromised or is insecure.

Similarly, I believe we can't claim `rclcpp` is insecure if I can `system(..)` a new ROS 2 node with the enclave of another.

(perhaps the use of files on a file system to store and distribute those certificates is a weak point? But again: not a security researcher)

Again: I'm not claiming there isn't something to fix here, but I believe it would be good to be clear about expectations and 'promises'. Perhaps you (@LichenXia198) could clarify what gave you the impression 'child' nodes exist and they can/should be compared to OS processes (with the implied parent-child relationship and related security implications).

---

PS: I'm just an interested observer here. I'm not a maintainer nor a contributor to the packages you've investigated.