ros2/ros2 #1590: Please add a LICENSE page

gmacario, 2024-08-07:
## Bug report

I could not find a LICENSE page in this repository.

Could you please add one so we can know under which license will our PR be contributed against?

**Required Info:**

- Operating System: N.A.
  - 
- Installation type: N.A.
  - 
- Version or commit hash: rolling
  - 
- DDS implementation: N.A.
  - 
- Client library (if applicable): N.A.
  - 

#### Steps to reproduce issue

Inspect git repository

```

```

#### Expected behavior

I would expect to find a LICENSE file in the toplevel directory which clearly states the license which governs this code

#### Actual behavior

#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

clalancette, 2024-08-07:
> I would expect to find a LICENSE file in the toplevel directory which clearly states the license which governs this code

So we don't have a LICENSE in this repository, because there is no code hosted here.  All that is listed here is some metadata.  What specifically are you looking to contribute to?

gmacario, 2024-08-07:
> So we don't have a LICENSE in this repository, because there is no code hosted here.

Understood, but I am making the case because I noticed there are quite a few hundred forks of this repo (I came here from https://github.com/space-ros/space-ros), and not having a LICENSE in the upstream repo can create issues such as those described in https://docs.github.com/en/repositories/managing-your-repositorys-settings-and-features/customizing-your-repository/licensing-a-repository

> You're under no obligation to choose a license. However, without a license, the default copyright laws apply, meaning that you retain all rights to your source code and no one may reproduce, distribute, or create derivative works from your work. If you're creating an open source project, we strongly encourage you to include an open source license. The [Open Source Guide](https://opensource.guide/legal/#which-open-source-license-is-appropriate-for-my-project) provides additional guidance on choosing the correct license for your project.

clalancette, 2024-08-07:
> Understood, but I am making the case because I noticed there are quite a few hundred forks of this repo (I came here from https://github.com/space-ros/space-ros), and not having a LICENSE in the [...]

Yes, but my question is, what would the LICENSE cover?

gmacario, 2024-08-08:
> Yes, but my question is, what would the LICENSE cover?

My point is well explained in one downstream repo which already includes code: https://github.com/space-ros/space-ros/issues/185#issuecomment-2275186064

gavanderhoorn, 2024-08-09:
That seems a bit circular?

MichaelOrlov, 2024-08-22:
@clalancette A friendly ping to follow up on this issue.

clalancette, 2024-08-23:
After discussing this for a bit, I think I finally understand what you are asking for.  If I can rephrase, what you are asking for is for us to put a LICENSE file in this repository so that when others contribute actual code to a fork of the repository, they will automatically do so under whatever LICENSE they are inheriting from here.

We don't think that is a good idea, for a couple of reasons:
1.  Putting a LICENSE file at the top-level repository here might give the wrong impression that it is the LICENSE for all of ROS 2, which is emphatically not the case.  The ROS 2 core is a mixture of open-source licenses, mostly Apache 2 and BSD.  However, packages that are not in the core are free to use any license they want.  Regardless, we don't want to give the wrong idea that the LICENSE here covers all of ROS 2.
2.  This repository really is not intended to hold code.  The fact that forks of this repository are doing so is entirely out of our control, but we would encourage them not to do so.  If they still want to do that, then I'll suggest that those forks add in a LICENSE file covering the code that they are adding.

Because of those reasons, I'm going to go ahead and close this issue.  But feel free to keep commenting.

closed 2024-08-23 by clalancette