ros2/ros2 #1591: ros2 service echo is not recognised!

PeymanAmirii, 2024-08-07:
Hi. I use ROS2 Iron on Ubuntu 22. All the commands for topics and services mentioned in the tutorial work correctly except for `ros2 service echo`. The terminal shows the following error:
`invalid choice: 'echo' (choose from 'call', 'find', 'list', 'type')`

It seems that the choice `echo` is not recognised by ros2 iron for the command `ros2 service`.
This command has been discussed in the tutorial here:
https://docs.ros.org/en/iron/Tutorials/Beginner-CLI-Tools/Understanding-ROS2-Services/Understanding-ROS2-Services.html#ros2-service-echo

clalancette, 2024-08-07:
> Hi. I use ROS2 Iron on Ubuntu 22. All the commands for topics and services mentioned in the tutorial work correctly except for `ros2 service echo`. The terminal shows the following error: `invalid choice: 'echo' (choose from 'call', 'find', 'list', 'type')`

You are correct.  That part of the documentation should only be for Jazzy and Rolling, not for Iron.  I'll open a PR to remove that bit of it.  Thanks for the report.

PeymanAmirii, 2024-08-07:
Thanks for your quick reply. So how should we echo services in ros2 iron?

clalancette, 2024-08-07:
> Thanks for your quick reply. So how should we echo services in ros2 iron?

That is a feature only available in Jazzy and later.

clalancette, 2024-08-07:
> That is a feature only available in Jazzy and later.

Actually, I'm wrong.  There is a more manual way to do it; I'll update the documentation to show that.