ros2/ros2 #1592: No std::cin / std::getline input, when running C++ node with ros2 launch

oskarsVismanis, 2024-08-13:
## Bug report

`std::getline`/`std::cin` doesn't take any input and stays blocking when called in a node, that's launched using `ros2 launch`

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- Installation type:
  - binaries
- Version or commit hash:
  - Humble
- DDS implementation:
  - CycloneDDS
- Client library (if applicable):
  - c++, rclcpp, rclpy

#### Steps to reproduce issue

I have a custom node, set up similarly to this

```
rclcpp::init(argc, argv);
rclcpp::Node::SharedPtr node_;

rclcpp::executors::MultiThreadedExecutor executor;
executor.add_node(node_);
std::thread([&executor]() { executor.spin(); }).detach();

std::string query_str;
std::cout << "Enter something: " << std::endl;
std::getline(std::cin, query_str);
```

with a launch file like this
```
node = Node(
   package="pkg_name",
        executable="node_name",
        output="screen",
        parameters=[
            "use_sim_time": True,
        ],
    )
)

return LaunchDescription([edi_pick_place_node,])
```

#### Expected behavior

When launched with `ros2 launch` I would expect it to reach `std::getline`/`std::cin`, take my input from the terminal and continue with the program.

#### Actual behavior

In reality, when launched with `ros2 launch`, the program gets stuck on the input calls and never accepts any input from the terminal.

#### Additional information

Running the node with `ros2 run`, however, works, as the program works as intended. 

I also tried creating a separate thread for the process like so

```
auto input = [&query_str]() {
   std::cout << "Enter something: " << std::endl;
   std::getline(std::cin, query_str);
};

std::thread(input).join();
```

but the behavior is the same between `ros2 launch` and `ros2 run`.

clalancette, 2024-08-21:
This is basically by design.  The purpose of launch files it to launch multiple processes.  What happens when more than one process wants to use `stdin`?  There could be ways of handling it, but in general it is a bad idea to launch things that need input from `stdin` via launch.

oskarsVismanis, 2024-08-22:
> This is basically by design. The purpose of launch files it to launch multiple processes. What happens when more than one process wants to use `stdin`? There could be ways of handling it, but in general it is a bad idea to launch things that need input from `stdin` via launch.

Ok, that makes sense, thanks for the comment! Still, there is no way to work around this for cases where it is certain, that no other input would be taken, correct? For my case, using launch would just be more convenient, but not the worst thing in the world to live without.

sloretz, 2024-08-29:
Thanks for the issue! Closing as working as intended. 🧇

closed 2024-08-29 by sloretz

zxfever, 2025-02-11:
hi bro, i face the same error, have you solved it ?