ros2/ros2 #1602: Custom message arrays in Python service response not working

julianpopescu, 2024-09-11:
## Bug report

**Required Info:**

- Operating System:
  - 
- Installation type:
  - binaries
- Version or commit hash:
  - Jazzy
- DDS implementation:
  - default
- Client library (if applicable):
  - 

#### Steps to reproduce issue
I'm trying to populate a service response array in a Python coded service node and I'm getting an assertion violations from one of the IDL generated files. The service Service.srv definition is the following:
```
string req
---
my_interfaces/MyCustomResponse[] resp
```
Where MyCustomResponse.msg is defined as following:
```
int64 an_id
```
#### Expected behavior

#### Actual behavior
The runtime error is something like:

python3: /projects/my_proj/build/my_interfaces/rosidl_generator_py/my_interfaces/msg/_my_custom_response_s.c:53: my_interfaces__msg__my_custom_response__convert_from_py: Assertion `strncmp("my_interfaces.msg._my_custom_response.MyCustomResponse", full_classname_dest, 54) == 0' failed.

fujitatomoya, 2024-09-11:
@julianpopescu thanks for creating issue.

could you provide the complete minimal package including idl, client and server code to reproduce the issue.

gavanderhoorn, 2024-09-12:
@julianpopescu: can you show how you:

> try[..] to populate a service response array in a Python coded service node

?

I've seen this assert before when trying to assign a regular `list[int]` to `resp`, instead of a `list[my_interfaces/MyCustomResponse]`.

The code is basically checking whether the field is set to an object with `type(..) == my_interfaces.msg._my_custom_response.MyCustomResponse`.

julianpopescu, 2024-09-16:
My apologies, I had a mistake in the Python code that was instantiating the objects in the response. Everything works fine.

closed 2024-09-16 by julianpopescu