ros2/ros2 #1604: ros2 node list returns empty in Jazzy when the node runs in humble

karthiknit1, 2024-09-19:
## Bug report

**Required Info:**

- Operating System:
  - Windows 11
- Installation type:
  - Binary
- Version or commit hash:
  - Both Humble and Jazzy latest binary versions
- DDS implementation:
  - FastDDS
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue
Source ROS 2 Humble environment in command line 1 and source ROS 2 Jazzy environment in command line 2
In Humble, run the following command:
**ros2 run demo_nodes_cpp talker**
In Jazzy, run the following command:
**ros2 node list**

The above command in Jazzy returns empty output, but **ros2 topic list** and **ros2 run demo_nodes_cpp** listener works absolutely fine.

Whereas if I do the opposite, things works as expected, i.e.,
In Jazzy, run the following command:
**ros2 run demo_nodes_cpp talker**
In Humble, run the following command:
**ros2 node list**

The above command in Humble shows the "/talker" node running in Jazzy.

#### Expected behavior
ros2 node list in Jazzy should return the node running in Humble, because topic list and pub/sub communication are working fine.
#### Actual behavior
ros2 node list in Jazzy is returning empty if the node is running in Humble, this is incorrect information.

#### Additional information
This happens with both FastDDS and CycloneDDS

karthiknit1, 2024-09-19:
@clalancette @dirk-thomas - could you provide your insights into this issue, and is there any workaround

fujitatomoya, 2024-09-19:
@karthiknit1 thanks for creating issue.

Unfortunately, cross-distro communication is not officially supported with ROS 2. see more details for https://github.com/ros2/ros2_documentation/issues/3288

There could be many reasons why your expectation in this issue report does not work, but the thing is that API/ABI might be incompatible over the distribution, that cannot guarantee the runtime, even compilation.

clalancette, 2024-09-21:
Yes, exactly what @fujitatomoya said, we don't guarantee this compatibility.

Because of that, I'm going to close this out.  We'll still get the note into the documentation about this separately.

closed 2024-09-21 by clalancette

rayee-github, 2024-09-23:
I have been troubled by version compatibility issues recently. Since incompatibility is already a fact, **why not explicitly define it in the official ROS 2 announcements or documentation?(or already have)** Currently, discussions on this topic are still scattered across various forum posts, making it difficult for us to explain compatibility issues to downstream customers.
Alternatively, ROS 2 could directly handle communication isolation between different versions at the software layer, rather than the current situation where successful communication is a matter of luck, and failures are expected (with unpredictable severe errors).

fujitatomoya, 2024-09-23:
@rayee-github you are right about that. and we just merged https://github.com/ros2/ros2_documentation/pull/4736.

gavanderhoorn, 2024-09-23:
@rayee-github wrote:
> Alternatively, ROS 2 could directly handle communication isolation between different versions at the software layer, [..]

at least for 'all' DDS RMWs, it should be feasible to actually make them compatible.

See the previous discussions you've most likely already found.

We 'just' need someone to do the work, or sponsor it.

Either each RMW implements the same (de)serialisation and control flow by convention, or msg, srv and action (de)serialisation and handling at the RMW layer needs to be standardised.